wasp-os/wasp-os #232: Unable to use wasptool / connect to device from linux

rennard, 2021-07-23:
Not sure if I'm missing something obvious, I've been trying and failing to push an app to my devices (sealed or dev). I've tried an Ubuntu 20 VM and a linux mint cinnamon native install, both OSs can see the devices in the bluetooth menu but can't connect. Using wasptool fails to connect to anything at all, and I've not been able to try to specify a mac address as I don't know how to find the mac address through the wasp-os menu. Am I missing something here? Cheers for any help

Pittconnect, 2021-07-24:
@rennard 'I've tried an Ubuntu 20 VM ' does not support hardware unless you running PRO version of Windows that too with WSL2. I have not played with mint cinnamon native,  but being native it should not have problem communicating with the outside hardware I use Pi 3 and have no problem using wasptool to connect to the device, to get the mac address you need to have hcitool that comes with bluez https://snapcraft.io/install/bluez/mint than if you run 'sudo hcitool lescan' it should scan for mac address using bluetooth

rennard, 2021-07-26:
@Pittconnect thanks for this. Although this didn't work for me it did send me in the right direction, so much appreciated!

For future people who find this, there were two things that confused me:
- I had issues with hcitool either returning tons of mac addresses or nothing, which made narrowing down which mac address I wanted very difficult. 
- I've not been able to connect to pair to PineTime from and computer or phone, which I had figured was something being wrong with the bluetooth, but I was able to connect using gadgetbridge. What I'm guessing is that the watch doesn't advertise itself for pairing, but specific apps can. Basically this was a red herring and had made me wonder whether my bluetooth adapter was broken.

What I actually did to get this working was use gadget bridge to find the mac address, and verify that I could connect to the device, and then use that in wasptool.py.

closed 2021-07-26 by rennard

pebri86, 2021-07-27:
Maybe rfkill blocked the bluetooth
Try this first
`sudo rfkill unblock bluetooth`

rennard, 2021-07-27:
cheers, I tried various things like this. Ultimately I'm not sure what the issue was, I seemed to get further using `bt-device` to list devices, but then got a segfault trying to list info about either of my pinetime devices. Using gadgetbridge seems to be the most foolproof method I've found so far

drtonyr, 2022-01-24:
I've also got this issue.   bt-device sees my watch:

> think0 tonyr: bt-device --list
> Added devices:
> P8 (F2:61:81:68:9B:F2)

But ./tools/wasptool fails to find the device (both from the wasp-os-0.4.1 release and from the current source):

> think0 tonyr: ./tools/wasptool --rtc
> ERROR: Cannot find suitable wasp-os device
> ~~~
> Scanning...
> ~~~