wasp-os/wasp-os #233: Weather app crashes when run on pinetime (but not in sim), and wasp.system.weatherinfo contains nonsense

0100001001000010, 2021-07-28:
I have a Pinetime, and I'm trying to make the weather app work. My current setup uses Gadgetbridge and Weather Notification on my phone for the weather. The weather app as shipped crashes immediately upon receiving data from Gadgetbridge. Upon looking at the exception, it appeared that `draw.string` in `_draw_label` was being given an int instead of a string as its first argument. This was because `wasp.system.weatherinfo` was storing the temperature, wind speed, and humidity as ints. I modified `_draw` to do type conversion, and this prevented it from crashing.
```
    def _draw(self):
        """Redraw the updated zones."""
        if self._temp_changed:
            self._draw_label(str(self._temp), 54, 36)
        if self._hum_changed:
            self._draw_label("Humidity: "+str(self._hum), 160)
        if self._txt_changed:
            self._draw_label(self._txt, 12)
        if self._wind_changed:
            self._draw_label("Wind: "+str(self._wind), 120)
        if self._loc_changed:
            self._draw_label(self._loc, 200)
```

I am not yet making a pull request for this because I am not sure whether this would be the best way to address this problem, and the second issue might or might not be related.

I have also noticed that `wasp.system.weatherinfo` seems to contain nonsense, and because the app takes the info directly from there, it displays nonsense. The description(`wasp.system.weatherinfo.get('txt')`), humidity(`wasp.system.weatherinfo.get('hum')`), and location (`wasp.system.weatherinfo.get('loc')`) are correct. However, the temperature (`wasp.system.weatherinfo.get('temp')`) seems to be an absurdly high value that is constant for each city. For example, Toronto's "temperature" is always 291, New York's temperature is always 294, and Houston's temperature is always 298, Los Angeles's temperature is always 293, etc. Wind speed seems to always be either blank or a small number that doesn't appear to have anything to do with the wind (I've seen 4, 7, and 11 so far) and seems to be random.

In all cases, the information displays correctly in Weather Notification. The app also has no problems on the simulator, but the data displayed there does not match the data displayed on weather notification, so I think that might be irrelevant fake data.

0100001001000010, 2021-07-28:
Actually, it doesn't seem like whatever is being reported as the temperature is completely constant. When I woke up, my local "temperature" had gone down by 2 units. A few hours later, it was back where it was, and a few hours later it was one above.

0100001001000010, 2021-07-28:
Wow. I completely failed to notice that the temperature and wind speed are not nonsense. Temperature is always in kelvin and wind speed is always in km/hr regardless of Gadgetbridge's unit setting. My workaround is to subtract 273 from self._temp before converting it into a string to convert into celsius, then multiply it by 1.8 and add 32 if display in Fahrenheit is desired. Wind speed can be left as is or can be multiplied by 0.62137119 to convert to miles per hour. I am not yet making a pull request for this because I feel like it is a very bad workaround when the real issue is somewhere else.

taitberlette, 2021-07-29:
Hello! Thanks for the report. 

When I created the app, I didn't have GadgetBridge installed on my phone, so I used my custom Android app. The server for my app converts the temperature, wind speed, and other data to strings before sending them to the client and the watch. My app also picks the preferred units (based on the location) on the server. When I created the app (for both Android and wasp-os), I followed the protocol [here](https://www.espruino.com/Gadgetbridge), which didn't specify the type of the value of the units, so I guessed it would be pre-formatted.

Also, yes, the simulator data is fake. You can view the demo code [here](https://github.com/daniel-thompson/wasp-os/blob/master/wasp/boards/simulator/main.py#L16).

The problem isn't the device itself (not a hardware problem), it's how the wasp-os app handles the data.

I'll have a look at the code later to try to fix the problems.

taitberlette, 2021-07-29:
Hey!

I think this should be fixed by my new pull request [here](https://github.com/daniel-thompson/wasp-os/pull/234). Please let me know if this looks good and if it works well with GadgetBridge.

Thanks.

closed 2021-11-14 by 0100001001000010