wasp-os/wasp-os #237: Add bluetooth pairing for security

goffi-contrib, 2021-08-12:
Hi,

first thanks for this project, it's really neat to have a micropython watch and indeed the easy developer onboarding seems reached.

At the moment, the security in wasp-os is a concern though: because there is not authentication, any apps which can do BT and reach the watch can have full access to the python console. Furthermore, the MAC address can be probably spoofed, and notification can then be read (with potentially really sensitive ones, like authentication codes, or normally encrypted messages).

I'm not a BLE expert, but I think that adding pairing/bonding would be a good start to make the whole thing more secure. Would it be possible to implement it?

Thanks

Pittconnect, 2021-08-17:
@goffi-contrib Its not clear to me what are you trying to pair it with ?

goffi-contrib, 2021-08-18:
Hello @Pittconnect , thanks for the feedback.

I'm using a PineTime.

My concern is that currently any application accessing bluetooth can have a full micropython shell on my watch without any way to notice (but the bluetooth icon). What I would like, is that the shell is not accessible without a pairing (e.g. with a number appearing on the watch that must be entered on the requesting application, so one knows that an application try to access the watch, and it can't do it without user input).

I'm not a specialist of BLE, but I think there are a couple of built-in options (see https://www.kynetics.com/docs/2018/BLE_Pairing_and_bonding/ for instance).

daniel-thompson, 2021-08-19:
> Would it be possible to implement it?

It would be great to have somebody implement this but I don't think it will be easy, if only because this is often discussed in the upstream micropython project and progress has been very slow.

Starting point is going to be here: https://github.com/micropython/micropython/blob/master/ports/nrf/drivers/bluetooth/ble_drv.c#L1013 but you would need to read a lot of the Softdevice example code to formuate any kind of strategy on how to tackle this work.

plan5, 2022-04-07:
Would it be difficult to vibrate when a new device connects?

Then at least it wouldn't go unnoticed and one could turn of bluetooth temporarily.