wasp-os/wasp-os #238: Connect to MQTT Server

NicholasIbarra, 2021-08-13:
Hello, 

Is it possible to create an application using MQTT? I went through the micropython folder and didn't find anything related to it, but is supported by MicroPython. 

If possible, can you point me in the right direction?

Pittconnect, 2021-08-21:
Are you trying to use MQTT for messaging /Notification or something else.  Which watch are you using pinetime, P8 or something else with wasp-os installed?

NicholasIbarra, 2021-08-22:
Yes, just want to connect to an Mqtt server and receive messages from a subscribed topic. I’d gladly share my example app once I get it to work. 

And the device is the pinetime.

Pittconnect, 2021-08-23:
Notification currently is limited check out the pdf manual updated 6-24-21 and search for notifications https://wasp-os.readthedocs.io/_/downloads/en/latest/pdf/ Let me know if it helps if you looking for code samples its in pager.py https://github.com/daniel-thompson/wasp-os/search?q=notifications

NicholasIbarra, 2021-08-23:
Thanks, I'll review.

closed 2021-08-23 by NicholasIbarra