wasp-os/wasp-os #239: Issue with drawing string after draw.fill()

AxelRHD, 2021-08-16:
The following code only prints colons instead of the input text:

```import wasp
import icons
import fonts
import math

class Playground():
	NAME = 'PlayGr'
	ICON = icons.app
	
	def __init__(self):
		self.screen = 0 # 0 = Start, 1 = Clock, 2 = Menu1
	
	def foreground(self):
		wasp.system.bar.clock = True
		self._draw()
		
		wasp.system.request_event(wasp.EventMask.TOUCH |
                                  wasp.EventMask.BUTTON)
    
	def press(self,button,state):
		if state == 0:
			self._draw()
	
	def _draw(self):
		draw = wasp.watch.drawable
		
		draw.reset()
		draw.fill()
		
		if self.screen == 0:
			draw.fill(0x001F,0,34,h=104)
			draw.fill(0xF920,0,138)
			
			draw.reset()
			draw.set_color(0xFFFF,0x001F)
			draw.set_font(fonts.sans28)
			draw.string("Resume game",0,math.floor(34+(104-28)/2),width=240)
			
		
		
		# ~ wasp.system.bar.draw()
```

Without the blue and red draw.fill() it works. Am I wrong with my approach or is this a bug?

Pittconnect, 2021-08-18:
@AxelRHD Thanks for sharing this draw.Fill() How are you doing it. I run a wasptool --console and at the >> prompt This will directly run the code on the watch and I get memory allocation errors https://pastebin.pl/view/c37d9363
So it may be a memory issue which font does the above code work at your end. 
`
>>> import gc
>>> gc.mem_free()
5936
>>>`
So I removed some apps to get me more free memory
xit console using Ctrl-X.

>>> import gc
>>> gc.mem_free()
10688
>>>
And this time I stop at the line before the san28 fonts and I still got memory allocation Error https://pastebin.pl/view/07b58b69

daniel-thompson, 2021-08-18:
> EDIT: seems to be a problem with the bigger fonts. With the default fontsize it works

That's more or less it! To reduce flash storage cost, the larger fonts only contain enough characters to draw clocks and stopwatches. Specifically the character set for the larger fonts is limited to: 0123456789:+-

AxelRHD, 2021-08-18:
I see. So all characters become colons in the bigger fonts.

closed 2021-08-18 by AxelRHD