wasp-os/wasp-os #240: How to simulate and create projects using Windows

fellington, 2021-08-19:
Hi,

So I am new at developing watch applications, and thought wasp-os would be a great way to develop since I am familiar with python. However, I have been struggling due to issues with capability with modules and different versions. I have issues with a requirement (pycairo because of wheel issues), the make sim command (pysdl2 due to it not finding the module with the path provided) and trying to test the code on the device.

I've been using WSL as recommended by a previous post from patchie, followed the installation guide from the docs, and have been trying to follow the Hello World app example too. Eventually, I want to use wasp-os to program the pinetime watch into a BLE server and have it do signal processing from the paired BLE device, but I've been having major issues getting started. If anyone has any suggestions or clearer instructions on how to run wasp-os on WSL, I would be really thankful.

My python version is 3.9.5, and I have installed all but the pycairo requirements.

Pittconnect, 2021-08-21:
There were some issues with python 3.9 that daniel did fix in #211 And WSL is funny being windows 10 always installing automatic patches It is difficult to have a stable platform. I would recommend using a Raspberry Pi or an old desktop and install Linux on it  Once we have a stable OS it lot more easy to troubleshoot and debug. 
@fellington  And what do you mean **_program the pinetime watch into a BLE server_** You want other (paired) bluetooth device to upload the data directly to PineTime over the air (OTA) ?

charliehawco, 2021-12-23:
I agree with fellington - is there any way to get list of the requirements but include the version number that you know works with the code on github? That way, we don't need to mess around trying to find what modules work, we can just install those specific versions and be on our way

daniel-thompson, 2021-12-24:
I'm not interested in documenting known working versions without reports on a version skew problem on a distro released in the last two years or so. Putting in explicit version numbers into the docs just makes it hard for folks with an older distro. I'd be willing to consider patches to add minimum version numbers but only if they are correct (e.g. a real minimum not just what was tested).

However... if you just want to check the version number for each package we use for CI testing then you can get these from the CI logs:
https://github.com/daniel-thompson/wasp-os/actions?query=is%3Asuccess+workflow%3A%22wasp-os+binary+distribution%22+branch%3Amaster++

Click on the top entry, then click "Build" for the logs and navigate down to "Install Packages".

Of course for WSL what is really missing is the install documentation... patches for that from Windows users remain welcome if they want to help out other Windows folks.