wasp-os/wasp-os #245: Stuck on bootloader. Dfu fails with nrf connect and android.

ddcash, 2021-08-28:
I have so far bricked both of my new pinetime devices using the install method. Please help.

ddcash, 2021-08-28:
I flashed the mcu-bootloader and then I'm stuck. It starts gets to 1 or 2% the watch reboots and shows the start pinecone and Bluetooth logo at the bottom. The log shows there is a dfu command error.

daniel-thompson, 2021-08-29:
If you can get as far as 2% then the bootloader is working be it you are having trouble with the radio link. It would be worth experimenting with the number of packets and MBR size settings in nRF Connect to try and make the link more reliable. You could also try with a different Android device (or use the GNU/Linux DFU tools).

ddcash, 2021-08-29:
Oh my gosh, thank you!
I used an older phone and it uploaded just fine. Phew what a relief. I have restored both devices pinetime watches.

closed 2021-09-05 by daniel-thompson

lenikhilsingh, 2022-06-19:
I tried loading the firmware with some other phone. In my case, 100% data transfer gets completed but still there's no change in the watch. It remains the same as described above. Mine is a development watch.

SteveAmor, 2022-06-19:
Unfortunately, the latest version of NRFConnect (4.25.3) does not work correctly for PineTime DFU. 4.24.4 is known to work. You can try GadgetBridge installed using F-Driod.