wasp-os/wasp-os #248: Interface crashes when the name of an app is too long

fgaz, 2021-09-10:
To reproduce:

* create a dummy `IHaveALongNameApp`
* start the watch
* it will hang on `main.py`

fgaz, 2021-09-11:
NAME, and yes

fgaz, 2021-09-11:
And here's the stack trace from the simulator:

```
Traceback (most recent call last):
  File "/home/fgaz/src/wasp-os/wasp/wasp.py", line 519, in run
    self._tick()
  File "/home/fgaz/src/wasp-os/wasp/wasp.py", line 482, in _tick
    self._handle_touch(event)
  File "/home/fgaz/src/wasp-os/wasp/wasp.py", line 436, in _handle_touch
    if self.app.swipe(event):
  File "/home/fgaz/src/wasp-os/wasp/apps/launcher.py", line 46, in swipe
    self._draw()
  File "/home/fgaz/src/wasp-os/wasp/apps/launcher.py", line 86, in _draw
    draw_app(page[0],   0,   0)
  File "/home/fgaz/src/wasp-os/wasp/apps/launcher.py", line 79, in draw_app
    draw.string(app.NAME, x, y+120-30, 120)
  File "/home/fgaz/src/wasp-os/wasp/draw565.py", line 321, in string
    self.fill(bg, x, y, leftpad, h)
  File "/home/fgaz/src/wasp-os/wasp/draw565.py", line 153, in fill
    display.set_window(x, y, w, h)
  File "/home/fgaz/src/wasp-os/wasp/drivers/st7789.py", line 132, in set_window
    window[2] = xp >> 8
ValueError: byte must be in range(0, 256)
```

daniel-thompson, 2021-09-11:
To be honest this is somewhat expected. If a developer gives an app that renders more then 120 pixels wide then the simulator will report the problem so it can be fixed (ideally before you put the app anywhere near a real watch... but a watch can be recovered fairly easily too).

Probably should be fixed in the docs rather than the code if the above is suprising.

The reason for this is that more or less the *only* use for NAME is to show it in the launcher. Therefore if the launcher automatically truncated long strings instead of reporting the error as an exception then the app author might not realize the long name they have provided for the app is pointless ;-) .