wasp-os/wasp-os #251: wrap() does not always work with string(width=...)

fgaz, 2021-09-11:
works:

```python
draw.string(filename[:(draw.wrap("testwithareallylongname", 240)[1])], 0, 200)
```

crashes:

```python
draw.string(filename[:(draw.wrap("testwithareallylongname", 240)[1])], 0, 200, width=240)
```

trace:

```
[omitted]
    draw.string(filename[:(draw.wrap("testwithareallylongname", 240)[1])], 0, 200, width=240)
  File "/home/fgaz/src/wasp-os/wasp/draw565.py", line 321, in string
    self.fill(bg, x, y, leftpad, h)
  File "/home/fgaz/src/wasp-os/wasp/draw565.py", line 153, in fill
    display.set_window(x, y, w, h)
  File "/home/fgaz/src/wasp-os/wasp/drivers/st7789.py", line 132, in set_window
    window[2] = xp >> 8
ValueError: byte must be in range(0, 256)

drtonyr, 2023-09-24:
I was just about to open an issue with my own debug example then I found this one (from 2 years ago).

I just started on improving wrap(), then gave up because I was getting similar errors.   The problem isn't with wrap, it's much lower down.  The problem is that some calls to draw.fill() that have zero pixels fail because display.set_window() can't cope with zero pixels when x = 0.

st7789.py contains the code:
```
  xp = x + width - 1
  ...
  window[2] = xp >> 8
```
We've both managed to trigger a strange edge case.  When both x is zero and the window width is zero then xp is -1 and `-1 >> 8` isn't in the range (0,256).  To see this here is the guts of the code that fails:
```
>>> draw.string('testwithareallylong', 0, 200, width=240)
```
but just indent by one pixel and it works:
```
>>> draw.string('testwithareallylong', 1, 200, width=239)
```

Now, the question is where to fix it?   There seems to be two choices:

1. fill(bg, x=0, y=0, w=None, h=None) - simply return if w*h is zero, before set_window() is called, this works
2. set_window(x, y, width, height) - this is low level, I don't understand this code.  Using `xp = max(x + width - 1, 0` seems to work.

both apps/gallery.py and apps/game_of_life.py call set_window().   Is it reasonable to call set_window() with zero width or height and expect it to work?   I don't know.

So I've tracked the problem down and I have two fixes.  @daniel-thompson, if you have time to read this, you probably know exactly where it should be fixed.   My gut feeling is that we should fix this as low level as possible, so ` xp = max(x + width - 1, 0)`  is the right thing to do - but I understand fill() and I don't fully understand set_window()

Can @daniel-thompson or anyone else help me decide where to fix this.   I'm getting good with pull requests now, once we know what fix we want I can easily

drtonyr, 2023-10-08:
I'm going to close this now.  We have two fixes, #464 deals with this specific problem and #460 the general case.  I have confirmed that this now work on the simulator:
```
>>> s = "testwithareallylongname"
>>> draw = wasp.watch.drawable
>>> draw.string(s[:(draw.wrap(s, 240)[1])], 0, 200, width=240)
```
[image]

closed 2023-10-08 by drtonyr