wasp-os/wasp-os #253: Gets stuck on software check

joseph58tech, 2021-09-19:
When I was going to install Wasp os, I ran into the problem of the watch getting stuck on the screen saying main.py
[image]

daniel-thompson, 2021-09-19:
How did you install, what version did you install and have you run wasp-os previously in this device?

joseph58tech, 2021-09-19:
> How did you install, what version did you install and have you run wasp-os previously in this device?

This was the first time installing Wasp os, but now I have a bigger problem, I just tried to reload infiniTime and the watch is stuck in a boot loop, but I have been able to get it out of the boot loop once already

daniel-thompson, 2021-09-19:
That still leaves "how did you install?" and "what version did you install?".

joseph58tech, 2021-09-19:
I got infiniTime back and now that's boot looping to

joseph58tech, 2021-09-19:
> That still leaves "how did you install?" and "what version did you install?".

Sorry about that, I installed the latest version, and I Installed it using the official guide

daniel-thompson, 2021-09-19:
This is still rather lacking in information (there being more than one latest version, there being both binary and source releases and there being more than one install method in the install guide). You should always provide comprehensive versioning information when reporting bugs!

joseph58tech, 2021-09-19:
It was version 4.1, and I didn't actually follow the guide thoroughly, so that being said, the way I installed it was load the bootloader through infiniTime which was successfull, after that I loaded micropython.zip and than I got that problem, and just so you know I used nRFconnect