wasp-os/wasp-os #254: Misconfigured ADC?

fgaz, 2021-09-20:
I noticed that the issued fixed in https://github.com/JF002/InfiniTime/pull/483 in infinitime might affect wasp-os too, and might explain the weird battery readings I'm getting.

The relevant line in micropython should be:

https://github.com/micropython/micropython/blob/782d5b2e534c96f4668443efe37c0514e8fcfe91/ports/nrf/modules/machine/adc.c#L136

Since the problem is specific to the pinetime's configuration, a longer ACQTIME might not be accepted by upstream though.

I'll start by patching it locally and seeing if it improves my readings.

fgaz, 2021-09-25:
After doing a couple of charge cycle, I can say that the new readings make way more sense. The min and max voltages are aligned to what one would expect from a new battery.

The discharge curve I observed with ACQTIME=3 had a sharp drop at the beginning, followed by days of low voltage (this is only by looking at the battery icon, i did no formal tests), while with ACQTIME=40 it matches what I would expect from a lithium battery.

I'll prepare a patch for your fork of micropython.

kLabz, 2021-09-26:
I'm trying this on my watch because it made sense to me since I almost never see the first ~20% (80-100%) of my battery.
However I'm not sure it's good either; my watch is now at 100% while it doesn't seem like it should be. I'll keep an eye on it all day to see if it makes more sense soon.

fgaz, 2021-09-26:
That's because the current % algorithm (wasp/drivers/battery.py) assumes 4V is 100%, while with the new measurements it should be 4.2V.

closed 2021-10-30 by daniel-thompson