wasp-os/wasp-os #258: DFU keeps failing

rennard, 2021-09-24:
Basically it seems to be a lottery whether I can get a new build onto my Pinetime or not. 
When I first started developing it seemed to be more stable, however recently I have to restart my dfu command several times over before it'll even connect to the device. And then frequently it will hang and disconnect and force me to restart, as shown below. The command I'm using is 
`make -j `nproc` BOARD=pinetime all && ./tools/ota-dfu/dfu.py -z build-pinetime/micropython.zip -a E5:97:B8:05:DE:79 --legacy`

This is what it looks when it hangs:
`Begin DFU
[##########........................................] 19.4%  (61 of 314 kb)Exception at line 166: a bytes-like object is required, not 'str'`

Pittconnect, 2021-09-27:
@rennard I encounter a similar problem, where it failed at 20% every single time. What I found that my application that made call to libraries that was not part of wasp-os operating system. Similar error also show about but without the bytes-like object is required message when the myapp is large and will not fit into the pinetime memory of 512KB memory So in that case the fix is use myapp.mpy instead of myapp.py and you can create myapp.mpy  using command `sh$ ./micropython/mpy-cross/mpy-cross -mno-unicode -march=armv7m myapp.py`

rennard, 2021-10-01:
@Pittconnect thanks for your response, unfortunately my issue is not easily replicable in that manner.
Sometimes I'll have the above error where it hangs at a certain %, and when re-ran it moves past that %, and sometimes it fails at 'Waiting for INIT DFU notification'. For both of these situations I can make no code changes, and rerun, usually after several attempts it will eventually work.

daniel-thompson, 2021-10-01:
"Exception at line 166: a bytes-like object is required, not 'str'" is a python2 to python3 porting issue. If you can find the missing string conversion at line 166 (although who knows of what file?) then you might get a clearer error message from the tool.

rennard, 2021-10-01:
Thank you for the response. I've been making the (presumably naive) assumption that my code isn't ran during the build or dfu process? However if that isn't the case, I only have 1 custom file, which is doing something similar to heart.py. 
e.g.
`    def tick(self, ticks):
        t = machine.Timer(id=1, period=8000000)
        t.start()
        self.hr_subtick()
        wasp.system.keep_awake()

        while t.time() < 41666:
            pass
        self.hr_subtick()

        while t.time() < 83332: # line 166 
            pass
        self.hr_subtick()

        t.stop()
        del t
`

daniel-thompson, 2021-10-01:
I think you might have misunderstood.

Nothing in your code does get executed during the DFU process. It is just a binary file transfer directly into the nRF52832's built-in flash.

The python2-to-python3 porting issue somewhere in tools/ota-dfu/ .

rennard, 2021-10-01:
I thought that was the case but I'm pretty new to all of this! Do you have any tips on debugging this? This has been plaguing me for a while now, every time I need to iterate my app it takes about 10 attempts before it works, with no code changes whatsoever when it eventually works. Verbose mode also doesn't give me anything to go on

Pittconnect, 2021-10-02:
@rennard It depands on if you wish to use python2 or python3 wasp-os does will not properly anything below python3.6.9 so I have been using python 3.7.x and it has worked for me. On the terminal type in `which python3` if you type in `which python` it will give you python2 version   and you will know what version of python3 you have installed Then you decide if you want to port your python2 code to python3 https://docs.python.org/3/howto/pyporting.html  Alternatively you can tweak you code it may be just a few line here is a example https://link.springer.com/content/pdf/bbm%3A978-1-4302-2416-7%2F1.pdf

daniel-thompson, 2021-10-03:
> I thought that was the case but I'm pretty new to all of this! Do you have any tips on debugging this? This has been plaguing me for a while now, every time I need to iterate my app it takes about [...]

daniel-thompson, 2021-10-03:
> you decide if you want to port your python2 code to python3

@Pittconnect is discussed above, the reporter has not written any python2 code. The reason for the exception is and unfixed string conversion in ota-dfu/ .

Pittconnect, 2021-10-06:
@rennard  So the issue has been resolved has it ?  Please keep us posted !! 
@daniel-thompson  Thanks for the clarification !!  I was going by your explation "Exception at line 166" Which I have seen quite a few times not on dfu but also when there is no bluetooth driver support 
`dietpi@DietPi:~/wasp-os$ ./tools/wasptool --device EF:46:76:DB:A7:CC --push TODO.rst
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 166, in read_nonblocking
    s = os.read(self.child_fd, size)
OSError: [Errno 5] Input/output error`

Here the problem is there dietpi OS does not have driver however one can only know if the secondary message followed by 166 are listed.
Should I open a new issue on this one ? @daniel-thompson so should it be handle by dietpi forum ?

thiswillbeyourgithub, 2023-09-16:
I fixed the whole thing in #222  btw