wasp-os/wasp-os #273: Is it possible to track HR continuously without sleeping?

rennard, 2021-11-10:
Apologies if this is stated somewhere, but is it possible for an app to read HR/accelerometer data continuously without the screen turning off or with the screen turning off but the app still running in the background?

kLabz, 2021-11-10:
Currently not (for HR anyway), and from my understanding it would have several drawbacks:
- battery life would likely suffer from this
- iiuc it has to be checked at 24hz which is a little challenge in heart app itself, and could be harder to do in background
- I think infinitime have issues with HR in background interferring with BLE connection on pinetime watches; not sure if it would apply to wasp-os too

Not impossible but would need some work and probably some choices to be made to balance battery life vs data accuracy.

rennard, 2021-11-11:
Thanks for the response, I might have a poke around and see if I can get anything working / how it performs

closed 2021-11-11 by rennard

thiswillbeyourgithub, 2021-12-16:
Hi!

I hope you don't mind me coming to this thread to ask my question :)

I am extremely interested in sleep tracking (even rudimentary) and it is the primary reason I want to buy a pinetime. Given that I only know python and shell (albeit well), I am naturally drawn to wasp-os as, if I understand the ecosystem correctly, I can install wasp-os on my future pinetime, write a python script for it and run it like an app.

Idealy, this script would start an endless loop throughout the night and record accelerometer data, and ideally heart rate every so often.

To save battery and storage, the data would be logged at a low frequency, and the screen would appear black / be turned off.

**Given the answer above, I don't understand if my plan is feasible or not. I understand that having a script running in the background can be impossible on such device but why would turning brightness to "black" be impossible too?**

I hope I'm not making myself looking like a fool, I have **no** experiment on embedded system so it's a whole new paradigm for me... But I'm eager to contribute code if I succeed!

leofidus, 2021-12-20:
I think your plan is very feasible. Having your app turn the screen off while running isn't a problem (I think you can just call `watch.display.mute(True)`). I read "you can't do it in the background" more as "you can't do it while other apps are running (including the watch face)". I agree with that, but for your use case it should be acceptable to just run a sleeptracking app while you sleep.

So you would have an app that in an endless loop sleeps a couple seconds or even minutes, then records some accelerometer data, then some heartrate data, puts them somewhere, then loops. That seems entirely feasible, as long as you don't need too much storage (though just dumping data on your smartphone is also trivial with a Bluetooth Terminal app).

thiswillbeyourgithub, 2021-12-20:
Great news!

Thanks a lot, I will try and make it work as I should have some time in january.

I intend to document everything in a repo. I already created it and wrote some ideas. If anyone is interested in a sleep tracking app and don't want to duplicate efforts, head over to my repo and don't hesitate to open an issue :)

https://github.com/thiswillbeyourgithub/sleep_tracker_pinetime_wasp-os

Pittconnect, 2021-12-29:
@thiswillbeyourgithub  I have try to test the current ppg app with wasp-os and it runs for about 2 to 3 hours continuously that too if during the sleep you do not accidentally touch it during sleep. So yes battery life is an issue You may get 6 or even 8 hours if you are able to turn off the display and have it run in the background. Normally without the continuous app running the battery last for 10 to 14 hours max on PineTime with wasp-os.  Other potential problem you run into is the coding because there are limited resources on the watch you cannot import numpy as np like he has done it here using python [](https://github.com/Rinkesh-Kumar/Health-Band-using-MAX30102-and-Adxl335/blob/master/Health%20Band%20Beta%201.py ) So my question is how are you going to write a sleep tracking App without importing NumPy ?

thiswillbeyourgithub, 2021-12-29:
Hi!

Thank you for the interest.

The current battery life of 14 hours max on wasp os is because the OS is new right ? I was assuming that updates would in the end bring it to about the level of the pinetime os. Am I wrong?

Or are you talking about continuous heart rate monitoring? 

By that way what is the current idle battery life?

Anyway I was thinking about starting with getting the accelerometer data every 1s and HR data every 60s. That  should not eat at the battery right?

I am aware of the limitations and the fact that I won't be able to import numpy. My plan was to mainly code things from scratch. I am pretty confident that we can get usable results using smart heuristics and efficient coding, (if not directly implementing a random forest algorithm :D but I'm not optimistic about that one). And that would be  great way to dip my toes into the embedded software world because my current work has nothing to do with any of this (I am in medical school).

I was very happy for example to see how the morse coding app was coded : a binary tree coded into a string.

All remarks greatly appreciated :)

leofidus, 2021-12-29:
There are definitely opportunities in the OS to reduce power usage. Some of them presumably as design decisions to make it easier to write and debug applications, some just because nobody has gotten around to changing it. For example if I remember correctly the accelerometer is just always running in normal-power-100hz mode, even if no application is using it (you can configure it to run faster or slower, which impacts power draw, or even shut it off). On the other hand wasp-os will never be as power-efficient as infinitime, just because the micropython software stack eats more power than optimized C++.

> Anyway I was thinking about starting with getting the accelerometer data every 1s and HR data every 60s. That should not eat at the battery right?

The accelerometer data won't eat the battery. Reading it takes basically no energy and it's on anyways (and doesn't take much power), so you're only paying for having the processor running. For the HR data you have to consider how long each measurement takes, since you have to run the sensor for some time to get useful data. If you run it for 6s every 60s that means it's on 1/10 of the time, so maybe 11h runtime?

thiswillbeyourgithub, 2021-12-29:
Thanks for the answer.

I think I read somewhere that the current HR implementation was pretty slow and naive so I might take a look at optimizing it too if it appears to be a bottleneck.

Anyway I was thinking on relying only on accelerometer data to get sleep state and only store HR optionnaly for curious users. So it will not endanger the whole sleep tracking project if HR ends up problematic.

I will open a separate thread to inquire about expected battery life.

Pittconnect, 2021-12-29:
@leofidus a few months ago I build an HRV app and run it on the wasp-os simulator it's a good way to test the code if you do not need to test the battery or the code size you can copy past and have the code running in a few hours in the wasp-os simulator  that measures all the parameters describe here https://github.com/paulvangentcom/heartrate_analysis_python it takes 10 seconds exactly you can see from 1:13 in the video to 1:23 one cycle  to take all the reading we need to measure health I had uploaded video https://www.youtube.com/watch?v=FsX2-1JJF6o But it does not work on the watch because of low memory and even if I got the code working by reducing the modules and making the code more efficient the battery does not allow the code run for very long.

thiswillbeyourgithub, 2022-04-13:
Has anyone had the time to make a heart tracking app that would set an alarm every X minutes to silently record HR over 10s, store it to a daily file like the step tracker then go back to sleep?

If the watch is already doing something, then just wait an additional minute.

I would love to have that but am kinda swamped :)

That seems more useful than the current heart app IMO.

khimaros, 2022-11-10:
@Pittconnect have you uploaded this HRV app anywhere? i'm interested in adding support for HRV calculation natively in InfiniTime or wasp-os and this sounds like a great starting point.

thiswillbeyourgithub, 2022-11-10:
@khimaros  you might be interested in reading this PR : https://github.com/daniel-thompson/wasp-os/pull/363