wasp-os/wasp-os #276: readthedocs builds are failing

fgaz, 2021-11-20:
https://readthedocs.org/projects/wasp-os/builds/
https://readthedocs.org/api/v2/build/15302589.txt

```
[rtd-command-info] start-time: 2021-11-17T15:36:58.631770Z, end-time: 2021-11-17T15:36:59.365684Z, duration: 0, exit-code: 2
/home/docs/checkouts/readthedocs.org/user_builds/wasp-os/envs/latest/bin/python -m sphinx -T -E -b html -d _build/doctrees -D language=en . _build/html
Running Sphinx v1.8.5
loading translations [en]... done
making output directory...
building [mo]: targets for 0 po files that are out of date
building [html]: targets for 14 source files that are out of date
updating environment: 14 added, 0 changed, 0 removed
reading sources... [  7%] README
reading sources... [ 14%] TODO

Traceback (most recent call last):
  File "/home/docs/checkouts/readthedocs.org/user_builds/wasp-os/envs/latest/lib/python3.7/site-packages/sphinx/cmd/build.py", line 304, in build_main
    app.build(args.force_all, filenames)
  File "/home/docs/checkouts/readthedocs.org/user_builds/wasp-os/envs/latest/lib/python3.7/site-packages/sphinx/application.py", line 341, in build
    self.builder.build_update()
  File "/home/docs/checkouts/readthedocs.org/user_builds/wasp-os/envs/latest/lib/python3.7/site-packages/sphinx/builders/__init__.py", line 347, in build_update
    len(to_build))
  File "/home/docs/checkouts/readthedocs.org/user_builds/wasp-os/envs/latest/lib/python3.7/site-packages/sphinx/builders/__init__.py", line 360, in build
    updated_docnames = set(self.read())
  File "/home/docs/checkouts/readthedocs.org/user_builds/wasp-os/envs/latest/lib/python3.7/site-packages/sphinx/builders/__init__.py", line 468, in read
    self._read_serial(docnames)
  File "/home/docs/checkouts/readthedocs.org/user_builds/wasp-os/envs/latest/lib/python3.7/site-packages/sphinx/builders/__init__.py", line 490, in _read_serial
    self.read_doc(docname)
  File "/home/docs/checkouts/readthedocs.org/user_builds/wasp-os/envs/latest/lib/python3.7/site-packages/sphinx/builders/__init__.py", line 534, in read_doc
    doctree = read_doc(self.app, self.env, self.env.doc2path(docname))
  File "/home/docs/checkouts/readthedocs.org/user_builds/wasp-os/envs/latest/lib/python3.7/site-packages/sphinx/io.py", line 318, in read_doc
    pub.publish()
  File "/home/docs/checkouts/readthedocs.org/user_builds/wasp-os/envs/latest/lib/python3.7/site-packages/docutils/core.py", line 219, in publish
    self.apply_transforms()
  File "/home/docs/checkouts/readthedocs.org/user_builds/wasp-os/envs/latest/lib/python3.7/site-packages/docutils/core.py", line 200, in apply_transforms
    self.document.transformer.apply_transforms()
  File "/home/docs/checkouts/readthedocs.org/user_builds/wasp-os/envs/latest/lib/python3.7/site-packages/sphinx/transforms/__init__.py", line 90, in apply_transforms
    Transformer.apply_transforms(self)
  File "/home/docs/checkouts/readthedocs.org/user_builds/wasp-os/envs/latest/lib/python3.7/site-packages/docutils/transforms/__init__.py", line 171, in apply_transforms
    transform.apply(**kwargs)
  File "/home/docs/checkouts/readthedocs.org/user_builds/wasp-os/envs/latest/lib/python3.7/site-packages/sphinx/transforms/__init__.py", line 245, in apply
    apply_source_workaround(n)
  File "/home/docs/checkouts/readthedocs.org/user_builds/wasp-os/envs/latest/lib/python3.7/site-packages/sphinx/util/nodes.py", line 94, in apply_source_workaround
    for classifier in reversed(node.parent.traverse(nodes.classifier)):
TypeError: 'generator' object is not reversible

Exception occurred:
  File "/home/docs/checkouts/readthedocs.org/user_builds/wasp-os/envs/latest/lib/python3.7/site-packages/sphinx/util/nodes.py", line 94, in apply_source_workaround
    for classifier in reversed(node.parent.traverse(nodes.classifier)):
TypeError: 'generator' object is not reversible
```

Pittconnect, 2021-12-06:
@fgaz How are you installing the sphinx tool chain ? What operating system is it ? I using windows with virtualbox or WSL to run the Linux distro ? Which linux distro are you using ?

fgaz, 2021-12-07:
It isn't failing on my computer, it's failing on readthedocs (see the OP)

fgaz, 2021-12-07:
Oh, it looks like it's passing now https://readthedocs.org/projects/wasp-os/builds/15383554/

Maybe it was a problem on their end. Closing.

closed 2021-12-07 by fgaz