wasp-os/wasp-os #278: Colmi P8 (confirmed nRF) black screen

twomiletower, 2021-11-28:
Bought a P8 sometime last year to play around with and recently dug it out again.

Confirmed as being the Nordic chip inside and previously successfully used DaFlasher to install the DaBootloader and ATCWatch firmware.

Decided to try WaspOS too - flashed an OTA update of micropython.zip (p8 version). DaFlasher confirmed the update but then the watch gave me a completely black screen, no bootloader, no nothing.

My first assumption is that WaspOS has failed to start and put the watch in an unusable state, but the bootloader is intact - symptoms are a black screen, unresponsive to button presses, BLE radio seems to be off as it doesn't show up any more in DaFlasher or any other app.

My other assumption is if I let the watch battery go flat then it may be recoverable, but I don't know. Does this sound like it's bricked and unrecoverable? It's not a huge issue if it is because I'm pretty curious about opening the P8 up and reprogramming it via SWD anyway, but it'd be nice to know exactly how bad I messed up!

Thanks for your time.

yavulan, 2021-12-03:
@twomiletower In case you do have P8b MOY-TON5, have you tried the solution 
https://github.com/daniel-thompson/wasp-os/issues/222#issuecomment-944937466 ?

twomiletower, 2021-12-03:
@yavulan - alas not - the smartwatch doesn't even turn on at all. no bootloader screen, no nothing, doesn't show up on BLE. at this point I went and got myself an STLink so if you have access to teardown instructions to get at the programming header on the board I think that's the best approach at this point. debonding the glue is gonna be fun.

Pittconnect, 2021-12-25:
It happened to me once and I shorted the wire when I was doing SWD what I did was waiit for batteries to drain and it came back oon like it says here  [https://www.reddit.com/r/PINE64official/comments/qv8lu4/is_this_a_dead_pinetime/](url)