wasp-os/wasp-os #281: Wasp-OS bootloader errors out every time I try loading anything.

melstav, 2021-12-07:
I received a sealed PineTime watch a week or so ago, and I decided to give Wasp_OS a try. I had been running InfiniTime 1.7.1.

I downloaded wasp-os-0.4.1.zip from the releases page, unpacked it on my phone (Motorola x4) and used nRF Connect to install the build-pinetime/reloader-mcuboot.zip

That installed successfully, and the watch rebooted to show the white pine cone and the start prompt pointing at the button. Unfortunately, every time I try writing a zipfile to the watch, it errors out and the watch resets to the bootloader. 

It does not matter whether I'm trying to load micropython.zip, reloader-mcuboot.zip, or reloader-factory.zip. Every one errors out in very much the same way. I even tried loading using DaFlasher, with similar results. It also does not seem to matter whether I try using "Connect with preferred PHY" or "Connect with Autoconnect" in nRF Connect.

In the hope that it's useful, I'm going to append a logfile from nRF Connect. I'm hoping that there's something simple I'm missing or at least that it's possible to recover without cutting the case open to expose the programming port.

-------------------------------------------

nRF Connect, 2021-12-07
PineDFU (E6:C1:19:50:06:CB)
V 17:17:57.209 Connecting to E6:C1:19:50:06:CB...
D 17:17:57.209 gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, preferred PHY = LE 2M)
D 17:17:58.187 [Callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I 17:17:58.187 Connected to E6:C1:19:50:06:CB
D 17:17:58.187 [Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
V 17:17:58.203 Discovering services...
D 17:17:58.203 gatt.discoverServices()
I 17:17:58.635 Connection parameters updated (interval: 7.5ms, latency: 0, timeout: 5000ms)
D 17:17:58.884 [Callback] Services discovered with status: 0
I 17:17:58.884 Services discovered
V 17:17:58.890 Generic Access (0x1800)
- Device Name [R W] (0x2A00)
- Appearance [R] (0x2A01)
- Peripheral Preferred Connection Parameters [R] (0x2A04)
- Central Address Resolution [R] (0x2AA6)
Generic Attribute (0x1801)
- Service Changed [I] (0x2A05)
   Client Characteristic Configuration (0x2902)
Device Firmware Update Service (00001530-1212-efde-1523-785feabcd123)
- DFU Packet [WNR] (00001532-1212-efde-1523-785feabcd123)
- DFU Control Point [N W] (00001531-1212-efde-1523-785feabcd123)
   Client Characteristic Configuration (0x2902)
- DFU Version [R] (00001534-1212-efde-1523-785feabcd123)
Device Information (0x180A)
- Manufacturer Name String [R] (0x2A29)
- Model Number String [R] (0x2A24)
- Firmware Revision String [R] (0x2A26)
D 17:17:58.890 gatt.setCharacteristicNotification(00002a05-0000-1000-8000-00805f9b34fb, true)
I 17:17:58.994 Connection parameters updated (interval: 45.0ms, latency: 0, timeout: 5000ms)
V 17:18:12.934 [DFU] DFU service started
V 17:18:12.934 [DFU] Opening file...
I 17:18:12.961 [DFU] Firmware file opened successfully
V 17:18:12.961 [DFU] Connecting to DFU target...
D 17:18:12.962 [DFU] gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, preferredPhy = LE_1M | LE_2M)
I 17:18:12.967 [DFU] Connected to E6:C1:19:50:06:CB
V 17:18:12.972 [DFU] Discovering services...
D 17:18:12.972 [DFU] gatt.discoverServices()
I 17:18:12.978 [DFU] Services discovered
D 17:18:13.004 [DFU] wait(1000)
V 17:18:14.009 [DFU] Reading DFU version number...
D 17:18:14.009 [DFU] gatt.readCharacteristic(00001534-1212-efde-1523-785feabcd123)
I 17:18:14.072 [DFU] Read Response received from 00001534-1212-efde-1523-785feabcd123, value (0x): 08-00
A 17:18:14.072 [DFU] Version number read: 0.8
D 17:18:14.075 [DFU] wait(1000)
V 17:18:15.076 [DFU] Enabling notifications for 00001531-1212-efde-1523-785feabcd123
D 17:18:15.076 [DFU] gatt.setCharacteristicNotification(00001531-1212-efde-1523-785feabcd123, true)
D 17:18:15.079 [DFU] gatt.writeDescriptor(00002902-0000-1000-8000-00805f9b34fb, value=0x01-00)
I 17:18:15.153 [DFU] Data written to descr.00001531-1212-efde-1523-785feabcd123, value (0x): 01-00
V 17:18:15.154 [DFU] Notifications enabled for 00001531-1212-efde-1523-785feabcd123
A 17:18:15.155 [DFU] Notifications enabled
D 17:18:15.155 [DFU] wait(1000)
V 17:18:16.156 [DFU] Writing to characteristic 00001531-1212-efde-1523-785feabcd123
D 17:18:16.157 [DFU] gatt.writeCharacteristic(00001531-1212-efde-1523-785feabcd123)
I 17:18:16.233 [DFU] Data written to 00001531-1212-efde-1523-785feabcd123, value (0x): 01-04
A 17:18:16.233 [DFU] DFU Start sent (Op Code = 1, Upload Mode = 4)
V 17:18:16.233 [DFU] Writing to characteristic 00001532-1212-efde-1523-785feabcd123
[... 52 lines cut ...]
D 17:18:25.402 [Callback] Services discovered with status: 0
I 17:18:25.402 Services discovered
V 17:18:25.420 Generic Access (0x1800)
- Device Name [R W] (0x2A00)
- Appearance [R] (0x2A01)
- Peripheral Preferred Connection Parameters [R] (0x2A04)
- Central Address Resolution [R] (0x2AA6)
Generic Attribute (0x1801)
- Service Changed [I] (0x2A05)
   Client Characteristic Configuration (0x2902)
Device Firmware Update Service (00001530-1212-efde-1523-785feabcd123)
- DFU Packet [WNR] (00001532-1212-efde-1523-785feabcd123)
- DFU Control Point [N W] (00001531-1212-efde-1523-785feabcd123)
   Client Characteristic Configuration (0x2902)
- DFU Version [R] (00001534-1212-efde-1523-785feabcd123)
Device Information (0x180A)
- Manufacturer Name String [R] (0x2A29)
- Model Number String [R] (0x2A24)
- Firmware Revision String [R] (0x2A26)
D 17:18:25.420 gatt.setCharacteristicNotification(00002a05-0000-1000-8000-00805f9b34fb, true)
I 17:18:25.501 Connection parameters updated (interval: 45.0ms, latency: 0, timeout: 5000ms)
D 17:23:25.981 [Callback] Connection state changed with status: 8 and new state: DISCONNECTED (0)
E 17:23:25.981 Error 8 (0x8): GATT CONN TIMEOUT
I 17:23:25.981 Disconnected
D 17:23:26.061 [Broadcast] Action received: android.bluetooth.device.action.ACL_DISCONNECTED

Pittconnect, 2021-12-25:
Infinite time 1.7.1 came after wasp-os 0.4 so it may not allow you to downgrade unless you are using reloader-factory.zip first to erase the new InfiniteTime 1.7.1 boot loader.  This will load the inifiniteTime bootloader not sure about the exact version Once you have that you can upload the reloader-mcuboot.zip that you trying to upload using OTA

daniel-thompson, 2021-12-26:
The version of Infinitime is unlikely to make any difference. Once you have the white pinecone then the watch doesn't have infinitime on it anymore!

Disconnection problems can often be solved by changing the "Number of packets" in nRF Connect -> Settings -> DFU options. Try reducing it signiciantly.

drtonyr, 2022-01-13:
> Disconnection problems can often be solved by changing the "Number of packets" in nRF Connect -> Settings -> DFU options. Try reducing it signiciantly.

This was incrededibly helpful advice, thanks a lot.  I have a P8 that was running a custom ATCwatch (via AtcWatchBuilder) but I love coding in Python.  bootloader-daflasher.zip went in fine but micropython.zip bombed out after the first packet or two.   I thought I'd softbricked my watch, but your fix (above) got me micropython.zip loaded, which is very exciting!