wasp-os/wasp-os #282: Cannot make simulator

charliehawco, 2021-12-24:
Hi all - I've downloaded the code and cloned it into a conda environment with the latest python, but when I try to create the simulator it throws an error?

```
C:\Users\sasdifsfjldkfsucnjksnskjed\Documents\wasp-os-master\wasp-os-master\wasp\wasp-os> make sim
MAKE Version 5.41  Copyright (c) 1987, 2014 Embarcadero Technologies, Inc.
Error makefile 15: Command syntax error
Error makefile 17: Command syntax error
Error makefile 102: Command syntax error
Error makefile 104: Command syntax error
Error makefile 126: Redefinition of target 'k9'
Error makefile 126: Redefinition of target 'p8'
*** 6 errors during make ***
```

charliehawco, 2021-12-24:
Ok - I have found your explanation for how to create the simulator (https://github.com/daniel-thompson/wasp-os/issues/218) and have managed to get to this point:

```
BLAHBLAHBLAH:~/wasp-os$ make sim
PYTHONDONTWRITEBYTECODE=1 PYTHONPATH=.:wasp/boards/simulator:wasp \
python3 -i wasp/boards/simulator/main.py
Traceback (most recent call last):
  File "wasp/boards/simulator/main.py", line 4, in <module>
    import wasp
  File "/home/sysadmin/wasp-os/wasp/wasp.py", line 18, in <module>
    import machine
  File "/home/sysadmin/wasp-os/wasp/boards/simulator/machine.py", line 9
    print(f'{self.__class__.__name__}.__init__{args} {kwargs}')
                                                             ^
SyntaxError: invalid syntax
>>> sudo make sim
  File "<stdin>", line 1
    sudo make sim
            ^
SyntaxError: invalid syntax
>>> 

```

daniel-thompson, 2021-12-24:
Can you try `python3 --version`? The simulator uses syntax features that were introduced in python 3.6.

charliehawco, 2021-12-24:
ah it's set as 3.5.3 - is there a particular version that you know it works with?

charliehawco, 2021-12-24:
ok so debian on wsl has decided it doesn't want to upgrade python

Pittconnect, 2021-12-25:
@charliehawco If you are using Debian buster Distro try this
`sudo apt install \
  wget git build-essential libsdl2-2.0-0 python3-click python3-gi \
  python3-numpy python3-pexpect python3-pil python3-pip python3-pydbus \
  python3-serial unzip
pip3 install --user cbor pysdl2`

daniel-thompson, 2021-12-26:
> If you are using Debian buster Distro

That's about the size of it... you need to install libsdl2-2.0.0 .

closed 2022-01-01 by charliehawco