wasp-os/wasp-os #286: Draw585

charliehawco, 2022-01-01:
Hi all - I'm wanting to modify the Draw585 module so that I have to specify the text colour, but because many of the modules use the self._bgfg, I'm unsure how to do it -

Below is a basic modification of one of the classes to show what I'm looking to do.

So, instead of:
```
    def string(self, s, x, y, width=None, right=False):
        print('s = {},x = {}, y = {}, width = {}, right = {}'.format(str(s),str(x),str(y),str(width),str(right)))
        """Draw a string at the supplied position.

        :param s:     String to render
        :param x:     X coordinate for the left-most pixels in the image
        :param y:     Y coordinate for the top-most pixels in the image
        :param width: If no width is provided then the text will be left
                      justified, otherwise the text will be centred within the
                      provided width and, importantly, the remaining width will
                      be filled with the background colour (to ensure that if
                      we update one string with a narrower one there is no
                      need to "undraw" it)
        :param right: If True (and width is set) then right justify rather than
                      centre the text
        """
        display = self._display
        bgfg = self._bgfg
        font = self._font
        bg = self._bgfg >> 16

        if width:
            (w, h) = _bounding_box(s, font)
            if right:
                leftpad = width - w
                rightpad = 0
            else:
                leftpad = (width - w) // 2
                rightpad = width - w - leftpad
            self.fill(bg, x, y, leftpad, h)
            x += leftpad

        for ch in s:
            glyph = font.get_ch(ch)
            _draw_glyph(display, glyph, x, y, bgfg)
            x += glyph[2] + 1

        if width:
            self.fill(bg, x, y, rightpad, h)
```

I'd like to be able to specify the foreground colour in the same method - for example, the chunk of code:
```
        for ch in s:
            glyph = font.get_ch(ch)
            _draw_glyph(display, glyph, x, y, bgfg)
            x += glyph[2] + 1
```

I'd like to be able to swap out the bgfg bit and change it to something like
```
        for ch in s:
            glyph = font.get_ch(ch)
            _draw_glyph(display, glyph, x, y, bg, fg)
            x += glyph[2] + 1
```

This way I can be sure that the colour I've set is used for that particular shape or bit of text - using the function that sets the foreground and background colour is awkward, especially if you want to set a particular text colour, but leave the background 'transparent'

charliehawco, 2022-01-01:
sorry, that was really badly written - basically, I'm very OCD when it comes to programming - and I've found that if I have to specify the font colour et al prior to running the method that draws the text, that all the text comes out in that colour - what I want to have is instead of

```
        draw.set_font(fonts.sans24)
        draw.set_color(wasp.system.theme('mid'))
        draw.string(t, 0, y + (i*24), 240)
```

In this particular method, I want to be able to specify the foreground colour specified in the draw.string() method

daniel-thompson, 2022-01-04:
The major contraints here is that:

- The _bitblit() method requires bgfg to be packed into a single argument. This is a micropython code generation contraint: viper methods cannot have more than four arguments. However the various functions that eventually call _bitblit() could have their arguments modified.
- Any changes must not regress string drawing performance (as measured on real hardware using the test app). The string drawing code is already pretty slow so performance wins out over athstestics here!