wasp-os/wasp-os #288: Can't connect via Bluetooth after initial Bootloader install

turnerendlesslearning, 2022-01-05:
Hi, I have a PineTime, about 3 months old.

I have connected it to my Linux Desktop using Siglo, and to my Android phone using GadgetBridge.

I used Siglo to install the bootloader and it seemed to work fine (shows the little pinecone, the word start with an arrow/line pointing towards the button).

Right after, though, it disconnected from Bluetooth, and it won't reconnect via phone or desktop. My computer recognizes the watch exists (listed as PineDFU in the Bluetooth settings), but can't connect to it.

If I press the button on the device, it goes black for a second, then shows the pinecone, and then the "start" + arrow again... Oh, also, there is a white bluetooth logo in the lower right corner of the screen.

Long press doesn't do anything different than a short press.

Now, I think the battery is dead: it goes to a black screen and doesn't respond to any button presses. If I just touch it for a second to the charger, it wakes up again briefly, but same symptoms.

Any suggestions? No rush. And I don't mind unsealing the back to fiddle with it... I'm not really using it in any situations where water resistance is an issue.

Thanks!

turnerendlesslearning, 2022-01-06:
Probably not helpful, but I'm going to try fully charging and see if it makes a difference. I've had a Garmin watch that paired poorly when the charge was low.

daniel-thompson, 2022-01-09:
Sounds like siglo might not have support for wasp-os. You can use dfu.py (in the wasp-os tools/ dir) or nRF Connect for Android for the next steps.

turnerendlesslearning, 2022-01-10:
Working perfectly! Thanks.

I'm still mid-installation process, but with nRF, it connected immediately and is uploading micropython.zip smoothly. Thanks again!

closed 2022-01-10 by turnerendlesslearning