wasp-os/wasp-os #289: Cannot flash using DFU-OTA on Ubuntu

jollyblondgiant, 2022-01-09:
Hello PIneOs wizards, 
I am blocked trying to flash wasp-os onto my pinetime running Infinitime. I received the closed watch from the dev kit as a gift. I am using Ubuntu and am able to connect via bluetoothctl and set the time, but am not able to flash wasp-os.
I have already followed the tutorials [here](https://wasp-os.readthedocs.io/en/latest/install.html#building-wasp-os-from-source) and [here](https://wiki.pine64.org/wiki/Switching_your_PineTime_between_InfiniTime_and_Wasp-os) looked everywhere I could think of for "Can't connect to device" and "Couldn't connect, will try DFU MAC" and "Need to switch to DFU mode" but alas, my kung fu is not strong. 

`$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.3 LTS
Release:	20.04
Codename:	focal

$ cd ~/wasp-os
$ ls build-pinetime/
bootloader-daflasher.zip  bootloader.hex  micropython.zip  reloader-factory.zip  reloader-mcuboot.zip  reloader.zip
$ sudo hcitool lescan
LE Scan ...
C3:7F:47:5F:08:1F InfiniTime
[... & others]
[in another tab, I connect and pair to pinetime with bluetoothctl]:
# info  C3:7F:47:5F:08:1F
Device C3:7F:47:5F:08:1F (random)
	Name: InfiniTime
	Alias: InfiniTime
	Appearance: 0x00c2
	Paired: yes
	Trusted: yes
	Blocked: no
	Connected: yes
	LegacyPairing: no
	UUID: Vendor specific           (00000000-78fc-48fe-8e23-433b3a1942d0)
	UUID: Vendor specific           (00001530-1212-efde-1523-785feabcd123)
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	UUID: Immediate Alert           (00001802-0000-1000-8000-00805f9b34fb)
	UUID: Current Time Service      (00001805-0000-1000-8000-00805f9b34fb)
	UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
	UUID: Heart Rate                (0000180d-0000-1000-8000-00805f9b34fb)
	UUID: Battery Service           (0000180f-0000-1000-8000-00805f9b34fb)
	UUID: Alert Notification Serv.. (00001811-0000-1000-8000-00805f9b34fb)
	UUID: Vendor specific           (00010000-78fc-48fe-8e23-433b3a1942d0)
 tools/ota-dfu/dfu.py -z build-pinetime/micropython.zip -a  C3:7F:47:5F:08:1F --legacy
Sending file firmware.bin to C3:7F:47:5F:08:1F
Binary imge size: 335764
Binary CRC32: 1060317325
Connecting to C3:7F:47:5F:08:1F
Couldn't connect, will try DFU MAC
Connecting to C3:7F:47:5F:08:20
Exception at line 164: Can't connect to device
[then i restart pinetime by holding the button for ~5s and releasing as soon as the pinecone graphic appears]
$ tools/ota-dfu/dfu.py -z build-pinetime/micropython.zip -a  C3:7F:47:5F:08:1F --legacy
Sending file firmware.bin to C3:7F:47:5F:08:1F
Binary imge size: 335764
Binary CRC32: 1060317325
Connecting to C3:7F:47:5F:08:1F
Need to switch to DFU mode
Exception at line 154: UUID not found: 00001531-1212-efde-1523-785feabcd123
[make sure I'm connected in the bluetoothctl tab]:
# info  C3:7F:47:5F:08:1F
Device C3:7F:47:5F:08:1F (random)
	Name: InfiniTime
	Alias: InfiniTime
	Appearance: 0x00c2
	Paired: no
	Trusted: yes
	Blocked: no
	Connected: no
	LegacyPairing: no
	UUID: Vendor specific           (00000000-78fc-48fe-8e23-433b3a1942d0)
	UUID: Vendor specific           (00001530-1212-efde-1523-785feabcd123)
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	UUID: Immediate Alert           (00001802-0000-1000-8000-00805f9b34fb)
	UUID: Current Time Service      (00001805-0000-1000-8000-00805f9b34fb)
	UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
	UUID: Heart Rate                (0000180d-0000-1000-8000-00805f9b34fb)
	UUID: Battery Service           (0000180f-0000-1000-8000-00805f9b34fb)
	UUID: Alert Notification Serv.. (00001811-0000-1000-8000-00805f9b34fb)
	UUID: Vendor specific           (00010000-78fc-48fe-8e23-433b3a1942d0)
# pair  C3:7F:47:5F:08:1F
Attempting to pair with C3:7F:47:5F:08:1F
[CHG] Device C3:7F:47:5F:08:1F Connected: yes
Request authorization
[agent] Accept pairing (yes/no): [CHG] Device C3:7F:47:5F:08:1F ServicesResolved: yes
[agent] Accept pairing (yes/no): yes
[CHG] Device C3:7F:47:5F:08:1F Paired: yes
Pairing successful
# connect  C3:7F:47:5F:08:1F
Attempting to connect to C3:7F:47:5F:08:1F
Connection successful
$ tools/ota-dfu/dfu.py -z build-pinetime/micropython.zip -a  C3:7F:47:5F:08:1F --legacy
Sending file firmware.bin to C3:7F:47:5F:08:1F
Binary imge size: 335764
Binary CRC32: 1060317325
Connecting to C3:7F:47:5F:08:1F
Couldn't connect, will try DFU MAC
Connecting to C3:7F:47:5F:08:20
Exception at line 164: Can't connect to device

`
What steps am i missing? why can't I connect to my device with dfu.py when I can connect and even set time from command line?
thank you, pineOs wizards, for hearing my plea

daniel-thompson, 2022-01-09:
dfu.py does not support Infinitime (although it could probably be added by somebody motivated) so you can't use dfu.py to do the initial switch. nRF Connect has support for Infinitime so if you have an Android phone that's the best way to switch (dfu.py does support wasp-os so after the initial switch you use dfu.py for everything else).

jollyblondgiant, 2022-01-09:
I don’t have an android phone. Can this be done with a pc running Ubuntu? I also have a raspi 4.

Pittconnect, 2022-01-11:
@jollyblondgiant  i would recommend Raspi 4 I used it works great Its good to build applications and debug using sims It gives the same look and feel that you can do on MacOS https://www.youtube.com/watch?v=tuk9Nmr3Jo8

fgaz, 2022-01-12:
IIRC infinitime apps such as siglo should work

JF002, 2022-01-12:
According to the logs, the MAC address of your watch is C3:7F:47:5F:08:1F, and the script tries to connect to C3:7F:47:5F:08:20 (MAC address + 1). I think this is expected for a device based on the NRF SoftDevice/bootloader (like wasp-os), but it's not compatible with InfiniTime.

However, InfiniTime is supported by multiple companion apps :
 - [Gadgetbridge](https://codeberg.org/Freeyourgadget/Gadgetbridge/) and [NRFConnect](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=fr&gl=US) on Android
 - [Amazfish](https://github.com/piggz/harbour-amazfish), [ITD](https://gitea.arsenm.dev/Arsen6331/itd) and [Siglo](https://github.com/alexr4535/siglo) on Linux
 - [InfiniLink](https://github.com/xan-m/InfiniLink) on iOS

All those apps should be able to send a DFU file to your PineTime running InfiniTime!

Pittconnect, 2022-01-14:
@JF002 That was helpful I have my sealed  It was working just fine with Wasp-os (python) Until I decided to switch to InfiniteTime(C++) using wasp-os/tools/ota-dfu/dfu.py -z reloader-factory.zip -a ,MAC-address --legacy IT seems for some reason to tool may not have completely uploaded the zip file and so I do not see the watch face after inifiniteTime logo or the firmware may be corrupt. Pressing the button down should normally take me to DFU move but it does NOT it just flickers like old TV screen static channel Here is the video https://youtu.be/-ArJuD9RVbI AT 0:06 in the video I plug in the charger and at 0:07 you see PineLogo (Large) From InfinitIme Which then fills up Green color at 0:14 and you see Infinite Logo at 0:15 and at 0:18 it powers OFF without me touching anything and starts the cycle again.  At that point it is Too late to press the button to get into DFU Mode SO I need to press the button at the time after the code turns green or before the cone turns green becaue I try holding the push button piror to plugging the cable and it does not work NOTICE in this video https://video.codingfield.com/videos/watch/5b70cc41-2e14-49cc-a631-0aa466957169 the InfiniteTime watch rolls back to 0.7.1 firmware I wish I knew how to do that any help is appreciated as it is a sealed PineTime watch and Opening it for SWD may not be worth it if the problem is with the charger!

JF002, 2022-01-14:
In theory, the DFU file should not be applied if the transfer is not complete. Also, the reloader checks the data written to the memory to ensure no error occured during the flashing of the new firmware.

I'm not sure, but in your video, your pinetime acts as if the battery was completely empty and your watch is reset by the brown-out protection (the MCU resets itself when the voltage is too low). Could you try with another power supply for the charger of the PineTime? It does not need a lot of power (~10mA, I think). A phone charger should be enough.

fgaz, 2022-01-14:
@Pittconnect that video looks exactly like what happens to mine when the battery is completely empty: it reboots at an increasingly rapid pace and due to this (?) it can't even _start_ charging (the charge icon just flashes briefly). It takes me multiple attempts sometimes over hours to get it to start charging continuously, so I usually avoid letting it run out of battery.

closed 2026-11-14 by mbeijen