wasp-os/wasp-os #291: "Chronometer" as default

Dibal, 2022-01-15:
Thanks for the nice Time-OS.

Problem: 
When I choose "Chronometer" (ie display with handles), it is only temporary. When I press the side button in order to switch display off -- display does not switch off but changes to the digital display.

Question:
* is it possible to choose face (ie chronometer <-> digital) in that way, that the side button does not switch it back?
* The display duration of "chronometer" is quite long. Is it possible to reduce the time of display (or enable the side button for switching it off?

Another comment: battery is running down after one day. Is it because BLE is running permanently?

fgaz, 2022-01-17:
> When I press the side button in order to switch display off -- display does not switch off but changes to the digital display.

Weird, this does not happen on my watch. What version are you using?

>  battery is running down after one day. Is it because BLE is running permanently?

That's one of the reasons. I get 4-5 days by disabling BLE using the DisaBLE app from the latest CI build.

Dibal, 2022-01-18:
> > When I press the side button in order to switch display off -- display does not switch off but changes to the digital display.
> 
> Weird, this does not happen on my watch. What version are you using?

The newest version I found on these pages: 0.4.1 (I did not find the option to ask the watch itself).

> 
> > battery is running down after one day. Is it because BLE is running permanently?
> 
> That's one of the reasons. I get 4-5 days by disabling BLE using the DisaBLE app from the latest CI build.

Can I switch BLE off as well (without affecting the bootloader)? Again I don't see any options on my watch

fgaz, 2022-01-18:
Hmm, that's weird. I'm not using that version, but when I did I remember it worked. Maybe try swiping left/right instead of pressing the button

> Can I switch BLE off as well (without affecting the bootloader)? Again I don't see any options on my watch

That's a new option (actually it's a separate app due to the fact that turning BLE back on requires a restart) that's not released yet. You can flash an unreleased version by downloading it from CI. Instructions are at https://wasp-os.readthedocs.io/en/latest/install.html#binary-downloads see the CI builds part. As written the warning, do NOT flash the bootloader from CI, flash only `micropython.zip`

Dibal, 2022-01-18:
I understand not to touch the bootloader. I find the latest CI here: https://github.com/daniel-thompson/wasp-os/actions/runs/1679807398; there are several zip files, but none named "micropython.zip". Do I have to build it myself? Is there a manual how to do?

fgaz, 2022-01-18:
You don't have to build it. You can find micropython.zip _inside_ one of those zip files (yes, it's a zip inside a zip). Pick the one that corresponds to your watch.

Dibal, 2022-01-18:
There are even two micropython.zip files included. In case you can give me a hint where I find a description I'd be happy. Otherwise this issue is closed, thank you!

daniel-thompson, 2022-01-19:
wasp-os runs on three different watches (Pine64 PineTime, Colmi P8 and Senbono K9). The different micropython.zip files are for the different devices.

daniel-thompson, 2022-01-19:
WIth the latest CI versions of wasp-os then you should have an all called Faces that can select the watch face. The selected face will be used until the next reset.

Dibal, 2022-01-20:
!!
What I did:
- download https://github.com/daniel-thompson/wasp-os/suites/4890351516/artifacts/140193362
- extract micropython.zip for pinetime
- reboot my watch (which already had a micropython bootloader)
- uploaded micropython.zip   via nRF-connect (Antroid)

What I got:
- Switch for faces which remain permanent even when I press side button
- Switch for turning BLE off
- New menu for alarm times, which allows setting of many alarm times, even with weekday choice

Simply great!

closed 2022-01-20 by Dibal