wasp-os/wasp-os #294: section `.text' will not fit in region `FLASH_TEXT'

drtonyr, 2022-01-24:
I'm making steady progress with wasp-os on my P8.  wasp-os-0.4.1 runs well and now I'm moving on to building from source:

  git clone https://github.com/daniel-thompson/wasp-os 
  cd wasp-os
  make submodules
  make softdevice
  make BOARD=p8 all
  ...
  LINK build-p8-s132/firmware.elf
  /usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: build-p8-s132/firmware.elf section `.text' will not fit in region `FLASH_TEXT'
  /usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: region `FLASH_TEXT' overflowed by 236 bytes
  collect2: error: ld returned 1 exit status
  make[1]: *** [Makefile:501: build-p8-s132/firmware.elf] Error 1
  make[1]: Leaving directory '/home/tonyr/wasp-os/micropython/ports/nrf'
  make: *** [Makefile:66: micropython] Error 2

Is this a known issue?  I would guess that I'm using a different compiler version or flags which results in more verbose code.

cgoates, 2022-01-25:
I had this issue as well.  Building clean from master is 460 bytes over for me.  I've built the firmware successfully for my P8 before without this issue.  If I go back two commits earlier than master (before the DisaBLE app was added), then it builds just fine, though I haven't tested that build on my watch yet.  Perhaps adding the DisaBLE app pushed it a bit over for the P8?

drtonyr, 2022-01-25:
Thanks, it's good to know I'm not alone.

For information, all three builds overflow for me:

pinetime overflowed by 396 bytes
p8 overflowed by 236 bytes
k9 overflowed by 492 bytes

The docker build has to be my next step, that should pull me closer to the tool chain that everyone else is using.

drtonyr, 2022-01-26:
I can confirm that when in the diocker shell everything compiles as it should.

```
CC build-p8-s132/pins_gen.c
LINK build-p8-s132/firmware.elf
   text	   data	    bss	    dec	    hex	filename
 335432	    172	   1844	 337448	  52628	build-p8-s132/firmware.elf
arm-none-eabi-objcopy -O binary build-p8-s132/firmware.elf build-p8-s132/firmware.bin
arm-none-eabi-objcopy -O ihex build-p8-s132/firmware.elf build-p8-s132/firmware.hex
make[1]: Leaving directory '/home/user/wasp-os/micropython/ports/nrf'
python3 -m nordicsemi dfu genpkg \
	--dev-type 0x0052 \
	--application micropython/ports/nrf/build-p8-s132/firmware.hex \
	build-p8/micropython.zip
Failed to import ecdsa, cannot do signing
Zip created at build-p8/micropython.zip
user@a8a6dfd72bc5:~/wasp-os$ 
```

cgoates, 2022-01-26:
Whelp, time to make another attempt at getting docker running, I guess.  Thanks for checking this out and posting the resolution.

SomethingGeneric, 2022-01-27:
^ for me on Pop 21.10 I've got a 9.2.1 toolchain and am also overflowing :(

fgaz, 2022-01-27:
Yeah you need version 10:

https://wasp-os.readthedocs.io/en/latest/install.html#install-prerequisites

> You will also need a toolchain for the Arm Cortex-M4. wasp-os is developed and tested using the GNU-RM toolchain (10-2020-q4) from Arm.

The supported compiler [was upgraded](https://github.com/daniel-thompson/wasp-os/commit/9012b85ede15c561ddb1998e39906548773dd988) from 9 to 10 exactly because 10 generates a bit smaller code and we're very close to the limit

drtonyr, 2022-01-28:
Thanks all.   I found https://askubuntu.com/questions/1243252/how-to-install-arm-none-eabi-gdb-on-ubuntu-20-04-lts-focal-fossa which says 'here is a script to generate a Debian package and install it to the /usr directory' (attached for convenience).   After sudo apt remove gcc-arm-none-eabi binutils-arm-none-eabi this script did the job and `make BOARD=p8 all` works as expected (outside of docker this time).

I think this matter is nicely closed.

`
#!/bin/bash

VER=15:10.3-2021.10-9
URL=https://developer.arm.com/-/media/Files/downloads/gnu-rm/10.3-2021.10/gcc-arm-none-eabi-10.3-2021.10-x86_64-linux.tar.bz2
echo "Creating gcc-arm-none-eabi debian package version $VER"

echo "Entering temporary directory..."
cd /tmp

echo "Downloading..."
curl -fSL -A "Mozilla/4.0" -o gcc-arm-none-eabi.tar "$URL"

echo "Extracting..."
tar -xf gcc-arm-none-eabi.tar
rm gcc-arm-none-eabi.tar

echo "Generating debian package..."
mkdir gcc-arm-none-eabi
mkdir gcc-arm-none-eabi/DEBIAN
mkdir gcc-arm-none-eabi/usr
echo "Package: gcc-arm-none-eabi"          >  gcc-arm-none-eabi/DEBIAN/control
echo "Version: $VER"                       >> gcc-arm-none-eabi/DEBIAN/control
echo "Architecture: amd64"                 >> gcc-arm-none-eabi/DEBIAN/control
echo "Maintainer: maintainer"              >> gcc-arm-none-eabi/DEBIAN/control
echo "Description: Arm Embedded toolchain" >> gcc-arm-none-eabi/DEBIAN/control
mv gcc-arm-none-eabi-*/* gcc-arm-none-eabi/usr/
dpkg-deb --build --root-owner-group gcc-arm-none-eabi

echo "Installing..."
sudo apt install ./gcc-arm-none-eabi.deb -y --allow-downgrades

echo "Removing temporary files..."
rm -r gcc-arm-none-eabi*

echo "Done."
`

closed 2022-01-28 by drtonyr