wasp-os/wasp-os #299: Is there a way to find the handle of a BLE characteristic under ubluepy?

rennard, 2022-01-31:


Pittconnect, 2022-01-31:
@rennard Is there a way for you to try a different laptop BLE connection depends on many variables one of them the device you are using to connect to the watch here I a reference https://github.com/daniel-thompson/wasp-os/issues/55 If it still has issues please specify if you are using P8 or Pinetime also what laptop device to make the connection call?

rennard, 2022-01-31:
Thanks for your response. Unfortunately I'm not able to test on a different laptop. Please could you explain how this might make a difference? I've been assuming the handles are just random IDs used and that I need to find a way to get a handle for a given characteristic, then hold onto it, such that I can compare them when I write/read is made. I'm using a PineTime

Pittconnect, 2022-02-01:
@jones139 Did you get the Bluetooth connection working [https://github.com/daniel-thompson/wasp-os/issues/207](url) I found this thread here https://forum.micropython.org/viewtopic.php?t=7009 may be useful

jones139, 2022-02-01:
@Pittconnect , no I did not get it working.  @rennard is working on the same problem that I was.

Pittconnect, 2022-02-02:
When I was in school they taught us to debug programming code problems using print line function in python, It is still common https://python.hotexamples.com/examples/mdb/-/getHandle/python-gethandle-function-examples.html 
Once @rennard has figured out the problem using Nordic Bluetooth conn_handler that is not included in wasp-os https://github.com/kylemanna/nordic-sdk-nrf5/blob/master/components/ble/common/ble_conn_state.h  perhaps he can create a blog about how to debug BLE connection issues as it can be very complex from timing issues to packet size to incorrect IRQ section like somebody has done it here for InfinitTime and PineTime https://infinitime.io/blog/2021-02/debug-ble/