wasp-os/wasp-os #300: [feature request] show Battery percentage

ITCactus, 2022-02-02:
I hope I haven't overlooked this, but i was not able to find the battery level in percentage on the watch itself.
i tried some time ago on PineTime, and i see the battery level indicator. but i can not find a screen where the percentage is hidden.

the second part is that companion app Gadgetbridge was not able to show this info. 
is this not implemented by the Gadgetbridge, or just wasp-os does not send this info?

fgaz, 2022-02-03:
> I hope I haven't overlooked this, but i was not able to find the battery level in percentage on the watch itself.

On the console, you can use `watch.battery.level()`

On the watch itself there is no percentage indicator.

> the second part is that companion app Gadgetbridge was not able to show this info.
is this not implemented by the Gadgetbridge, or just wasp-os does not send this info?

wasp-os does not send battery info yet:

https://github.com/daniel-thompson/wasp-os/blob/5f28d05cc093b54153a72860dc68af7f01dd4e19/wasp/gadgetbridge.py#L3-L19

thiswillbeyourgithub, 2022-02-24:
@fgaz, by any chance, could you point me to where I could take a look at gadgetbridge's side of this?

I am interested in sending local sleep data to gadget bridge but don't know the first thing about their project and even struggle to find the documentation.