wasp-os/wasp-os #319: UTF-8 support in the wasp-os widget library

TheFocusDevHUN, 2022-04-03:
Hello! 

I've just started experimenting with wasp-os using the simulator,  and I've discovered that the widget library doesn't like strings with non ASCII characters. Specifically I want to extend the calculator app a little bit to have multiple operations by implementing a "Shift" option maybe with a swipe, I don't know. Anyway when  I've tried to replace a button with a square root symbol, the button showed a question mark. So it turns out that wasp is only supporting ASCII at the moment. It would be nice to have UTF-8 support, especially when a non- English user wants to translate the system to their native language.

Thank you!

drtonyr, 2022-04-03:
I think you'll find that it's a design decision of micropython, not WASP.  Or that's what I found on the web when I had similar thoughts.

TheFocusDevHUN, 2022-04-03:
Maybe I've found the problem. 
https://github.com/peterhinch/micropython-font-to-py#32-limitations