wasp-os/wasp-os #322: Is anyone interested in a 'Four in a row' App?

drtonyr, 2022-04-09:
I wanted to code something 'smart' for my smartwatch and as 2048 and Game of Life were already done I chose Four in a Row.

There is a wonderful Kaggle community (https://www.kaggle.com/c/connectx) so I'm confident I have a good implementation of the perfect search in Python3.  I also have a basic wasp-os implementation of the graphics, no game play yet.   Combining the two will give basic graphics and perfect game play, which is my goal.

I'm writing this 'issue' to ask if anyone wants the code.  It was always just a personal project, but if anyone is interested you are welcome  to the code.  If there is serious interest then I'd be willing to work with others to make it wasp-os standard.   There will be a large hash table on /flash that may make it interesting/repulsive for some,  There will be work to be done in implementing a 'difficulty level', it's no fun losing to your £10 smartwatch every time.

I'm really not expecting anyone to care about my personal project, but if I don't mention it here then nobody will know.  It's fine if this 'issue' is immediatly marked as closed, at least it can be found by future users and they can get in touch if they want.

[image]

fgaz, 2022-04-11:
I don't have any authority here, but I think it would be welcome! After all there's unlimited space in [`apps/`](https://github.com/daniel-thompson/wasp-os/tree/master/apps)

drtonyr, 2022-04-12:
@fgaz Agreed, but as of now I don't see wasp-os/apps on my watch (it's fine with make sim) and I remember a thread which said that the Gallary app ran very slowly from there.   It's on my TO DO list to understand wasp-os/apps, both getting them to show up and any speed issues.

Pittconnect, 2022-04-14:
@drtonyr Hello Tony if you check line 42 to 54 these are all the apps that will show up
unless you modify software.py and add your 'Four in a row' App https://github.com/daniel-thompson/wasp-os/blob/master/wasp/apps/software.py

drtonyr, 2022-04-14:
Thanks, but that's what I'm using at present.  My understanding is this is the apps in wasp-os/wasp/apps and these are the ones that run fast but we are short on memory.  For example, I just got my first version of the code in this evening, 1750 bytes over, down to 1250 bytes over after a bit of hacking.

The alternative is wasp/apps which sits on /flash which is largely unused.  I believe code runs slower from here.  When I run 'make sim' I get app from this space, but not when I do the standard install to watch.  There may be a way of pushing stuff to /flash after the WASP install, I haven't found it yet.

For now I'm writing for the wasp-os/wasp/apps space, but if anyone could point me to the documentation for the larger slower space I'd give it a go.

drtonyr, 2022-04-15:
[ quote below moved to this thread ]
> @drtonyr It is true that wasp-os is based of micropython but you have to remember that when one runs to code on the watch the resources are very limited due to small hardware foot print and memory limitation. An App may run on the make sim because it all the resources are there on your computer. Check out 4.4.4 a binary format upload https://wasp-os.readthedocs.io/_/downloads/en/latest/pdf/ also 4.4.5 on how to freeze you app so it loaded every time on your watch. The four_in_row.py not call are references to other external libraries like nump.py if it does it will work on the sim but on on the watch as any external library call must be loaded into the micropython.zip prior to uploading to the watch.

Thanks @Pittconnect - that's very helpful.    I liked the way you gave me the PDF so that I could find the numberparagraph easily.

Thanks to your help, I've cracked it - it did take more than the documentation, but the magic incantation is:
```
for file in apps/*.py; do ./micropython/mpy-cross/mpy-cross -mno-unicode -march=armv7m $file ; done
for file in apps/*.mpy; do ./tools/wasptool --binary --upload $file --as $file ; done
```
This creates the mpy and puts them in the right place to be autodetected and so run like 'make sim'.  ReadMe.mpy works for me, but more than that and I'm getting 'Your watch is out of memory'.

It does seem that some reorganisation would help.   For example, can't we move the app icons to (say) /flash/icons?  This  would free up firmware space which is at a premium.

fgaz, 2022-04-15:
Another way to free up space would be #283

drtonyr, 2022-04-19:
So I've just got my implementation of Four In a Row to play a decent game.   I've gone for minimal size and I'm really pleased at the current 2363 bytes.   There really isn't much varaible usage either.  This seems to compare reasonably with the autoloaded apps - it's not the biggest by a fair way:

```
think0 tonyr: ls -lta wasp-os/apps/*mpy
-rw-r--r-- 1 tonyr tonyr 2363 Apr 19 10:01 wasp-os/apps/fourInARow.mpy
-rw-r--r-- 1 tonyr tonyr 1654 Apr 15 08:03 wasp-os/apps/Morse.mpy
-rw-r--r-- 1 tonyr tonyr  228 Apr 15 08:03 wasp-os/apps/ReadMe.mpy
-rw-r--r-- 1 tonyr tonyr 3698 Apr 15 08:03 wasp-os/apps/GameOfLife.mpy
```
It works fine when run from the firmware, when run from /flash it gives 'Your watch is low on memory.  You may want to reboot.' - even straight after a reboot.

Including my app under wasp-os/apps seemed the cleanest way to get it into the distrubution, however it doesn't work.  But there again neither do any of the other apps in there for me, except for the minimal ReadMe.mpy.

What's the best way forward?   Does wasp-os/apps work for other people?  Is wasp-os/apps doomed as it'll never run anything of a few kbytes in size?

fgaz, 2022-04-20:
> What's the best way forward? Does wasp-os/apps work for other people? Is wasp-os/apps doomed as it'll never run anything of a few kbytes in size?

I use it for a few apps. Morse does work for me, but I'm using a slightly older version so maybe it increased in size. I can also load #275 and #277, though only one at a time.

drtonyr, 2022-04-25:
> I use it for a few apps. Morse does work for me, but I'm using a slightly older version so maybe it increased in size. I can also load #275 and #277, though only one at a time.

Thanks again @fgaz.  I'm making progress.  After some work, then sometimes I can run my FourInARowApp from flash, sometimes not.  If I don't register apps.faces.FacesApp() on boot (L173 of wasp.py) then I can run.  So I may be able to be a bit tighter on my RAM (both mpy and variable space) and make it into the release this way.  I'm working on it.

It would be really interesting to know what is using the RAM on boot, I suspect there is quite a bit that could be reasonably reclaimed.

drtonyr, 2022-05-06:
I've just lost my P8 so now is a good time to pause/stop.   I'm really pleased with my Four In A Row app, it's only 2485 bytes (mpy) and so witll run from RAM.  It uses some really cool algorithms to do fast search so it's a real challenge to beat.  There is a info page that lets you chose the difficulty level and displays win stats at that level.   I think the only remaining bugs are that sometimes the top line doesn't refresh when swithcing away from the app and back to it.   It's a cut down version of the code that I submitted to the ConnectX Kaggle competition which was top of the leaderboard this morning.  It beats me on the top levels, and at 2485 bytes it's just what I wanted - a tiny efficiient smart app for a smart watch.

[FourInARow.txt](https://github.com/daniel-thompson/wasp-os/files/8640163/FourInARow.txt)

daniel-thompson, 2022-09-30:
Regarding interest in apps for wasp-os. As @fgaz said there are two folders in wasp-os. wasp/apps for default apps (e.g. those that are included in the image by default) and apps/ (for bonus apps, these appear in the docs and simulator but aren't included by default in the images for real hardware.

The idea is that the wasp-os repo is the best place to find apps... and that wasp-os is intended to be an OS where the user is connected to the source code (in other words we hope to attract users/developers who can change the defaults to their taste from there).

So yes, a pull request is encouraged.

drtonyr, 2022-10-01:
Thanks @daniel-thompson.  As you might have realised by my recent activity, my p8 was returned to me (I also bought a couple more P8b for good measure).

As you are well aware, there isn't much RAM for anything that goes in wasp-os/apps.  To encourage a lot of interesting apps to be written there should be a policy specifying how much RAM should be available for wasp-os/apps - then app writers would know what they could and could not do.  At the moment it's pretry frustrating, even a very small app may or may not run.   To really get a lot written implies more RAM being freed.

The other thing that would help is to state that apps should be interchangable between wasp-os/apps and wasp-os/wasp/apps - IIRC the major inconsistancy is in file naming convention, wasp/apps/ start with an upper case letter and wasp-os/wasp/apps lower.   Or, this could be taken a step further and all apps could be in wasp-os/wasp/apps and wasp/boards/manifest_240x240.py could list those that get automatically pushed into /flash/apps.

drtonyr, 2023-10-12:
closing this issue as now merged in #451

closed 2023-10-12 by drtonyr