wasp-os/wasp-os #323: wasp/apps/gallery.py uses 'except FileNotFoundError'

drtonyr, 2022-04-10:
On a clean install and with no images to view I get: NameError: name 'FileNotFoundError' isn't defined.

It works as expected with 'make sim', it appears to be a python3/micropython difference: https://forum.micropython.org/viewtopic.php?t=5125

drtonyr, 2022-04-11:
... to be clearer, as the micropython forum topic thread says, replacing FileNotFoundError by OSError 'works on both MicroPython and PC Python'
```
think0 tonyr: diff -c gallery.py-orig gallery.py
*** gallery.py-orig	2022-04-09 15:00:44.453279842 +0100
--- gallery.py	2022-04-11 07:40:30.541947175 +0100
***************
*** 63,69 ****
      def foreground(self):
          try:
              self.files = wasp.watch.os.listdir("gallery")
!         except FileNotFoundError:
              self.files = []
          # In case some images were deleted, we reset the index every time
          self.index = 0
--- 63,69 ----
      def foreground(self):
          try:
              self.files = wasp.watch.os.listdir("gallery")
!         except OSError:
              self.files = []
          # In case some images were deleted, we reset the index every time
          self.index = 0
```

fgaz, 2022-04-11:
Is there a way to match on the specific OSError that stands for file not found?

drtonyr, 2022-04-12:
I hadn't used micropython before WASP so I don't know.  I'd have thought it likely that if there was something better then someone would have improved the answer on the micropython forum.

Pittconnect, 2022-04-14:
@fgaz  Micropython uses Errno codes numbers that can be decoded using this link https://www.ibm.com/docs/en/zvse/6.2?topic=SSB27H_6.2.0/fa2ti_errno_values.html
@drtonyr It is true that wasp-os is based of micropython but you have to remember that when one runs to code on the watch the resources are very limited due to small hardware foot print and memory limitation. An App may run on the make sim because it all the resources are there on your computer. Check out 4.4.4 a binary format upload https://wasp-os.readthedocs.io/_/downloads/en/latest/pdf/ also 4.4.5 on how to freeze you app so it loaded every time on your watch. The four_in_row.py not call are references to other external libraries like nump.py if it does it will work on the sim but on on the watch as any external library call must be loaded into the micropython.zip prior to uploading to the watch.

closed 2026-11-13 by mbeijen