wasp-os/wasp-os #329: [Enhancement] Waypoint Application / Dynamic GPS

MLGHerobrine, 2022-04-29:
This idea may not work, but it's worth a try:

Using the accelerometer in the watch, you can track approximately how fast/far someone is moving. According to [this pdf](https://aatishb.com/materials/srr/workshop3.pdf), you can find the yaw/orientation of a person.
Theoretically, with this knowledge, you should be able to track a person's movements (I believe it's called Dynamic GPS). 

The idea of this app is that it would create a waypoint, and track the person's movements from that waypoint. That way, you could return to the waypoint with the following options:

1. Compass
Displays the direction to the waypoint from your current position, like a compass.
2. Map
Shows the waypoint relative to your location and orientation, like a map. Helpful for visualizing the distance between you and the waypoint.
3. Historical Map
Similar to a map, except that it also shows the path you took once you created the waypoint. Useful for retracing your steps in a scenario, like hiking. Possibly memory/storage heavy.

Overall, this would be a useful feature for sure. It wouldn't be 100% accurate, but if it works, it would be accurate enough.
As soon as I get my hands on a PineTime, I'll try making something like this.

SteveAmor, 2022-04-30:
With just an accelerometer and gyroscope, you can't determine which direction and speed something is moving to/from a waypoint, you can only work out the orientation of a static object. Here is a video of what is achievable with 6 degrees of freedom https://youtu.be/uTmZjBErJTc  
An acc/gyro compliments GPS to provide even more accurate tracking, and a compass adds even more accuracy.  
Alas, we only have an accelerometer in the pinetime which isn't much help in this use case and at best, will provide the rough orientation of a static object (accelerometers are noisy so the example video above would be a little shaky, it the addition of thr gyro that removes the acc noise.) But, it can be used for a game/robot controller 😀

MLGHerobrine, 2022-04-30:
Thank you for clearing that up! I don't have a PineTime yet, so it's very hard to test. I'll still try to make a proof of concept if I can when I get mine.

closed 2022-04-30 by MLGHerobrine