wasp-os/wasp-os #331: [feature] is it possible to add support for long press touch ?

thiswillbeyourgithub, 2022-05-08:
Or is there a hardware limitation ?

daniel-thompson, 2022-05-09:
Looks like the hardware can detect long press:
https://wiki.pine64.org/wiki/PineTime#Touch_panel

As the wiki says _"there is not much information about this chip on the internet"_ and I think there used to be even less when I was first hacking together the event system.

It's certainly worth adding a bit of code to log the event stream and decide how to distinguish between events. I think the problem might be that the single click event still occurs during a long-press and that there is no timeout event to confirm a single click (so whilst we can detect long press we might already have performed the UI action associated with short press first).

thiswillbeyourgithub, 2022-05-09:
After seeing [this line](https://github.com/daniel-thompson/wasp-os/blob/09af1440ef5af0dd928cf99990906d08accf2b27/wasp/drivers/touch.py#L39), am I correct in assuming that every 300ms the watchs polls for touch event ?

If so, wouldn't it work to cache the last event in an attribute in class TouchButton and add a test in self.get_event : if self.event[0] == self.latest_event: return long_touch_event or something?