wasp-os/wasp-os #332: how to send "data" over ble ?

thiswillbeyourgithub, 2022-05-08:
Hi,

I was imagining sending packets around via bluetooth, having them picked up by my raspberry pi and doing home automation stuff.

Is there a way to send data from the watch via ble without connection ? I did not find related things in the documentation nor in the ble module.

Thanks!

daniel-thompson, 2022-05-09:
It can be done but I would require C level changes to micropython (AFAIK having the REPL enabled over bluetooth disables most of the ble module capabilities since it provides its own set of GATT services).

thiswillbeyourgithub, 2022-05-09:
Thanks! I don't know C so I can't contribute it. Eager to play with it if this gets implemented!

thiswillbeyourgithub, 2022-08-21:
> It can be done but I would require C level changes to micropython (AFAIK having the REPL enabled over bluetooth disables most of the ble module capabilities since it provides its own set of GATT services).

Does that mean that turning the REPL off temporarily would "allow" for ble capabilities? I'm really looking for a way to transmit data, no matter how inelegant or hackish...

Btw related to #110 and #347