wasp-os/wasp-os #334: fetch raw accelerometer data from Pine64 to build various algorithms on it

CIDROY-Tek, 2022-05-25:
Greetings!

I'd like to fetch raw accelerometer readings from pinetime pine64 smartwatch to be able to build various algorithms on it. Could you please suggest some method to achieve the same?

fgaz, 2022-05-25:
#275 implements an api to do that, but I'll have to address @daniel-thompson's comments

thiswillbeyourgithub, 2022-05-25:
I don't know what you want to use this for but you can find an example of using @fgaz's API of my app SleepTk that tries to put the alarm clock at the best time depending on your sleep cycle (using heart rate and body movement).

This feature is not yet working btw but it's still a very good alarm clock :D 

Here's the link : https://github.com/thiswillbeyourgithub/sleep_tracker_pinetime_wasp-os