wasp-os/wasp-os #341: How to remeber paired phone between waspos flash

thiswillbeyourgithub, 2022-06-20:
Hi,

I very frequently update my wasp os on my pinetime. It's a pain to have to manually resync it to gadget bridge.

Is there currently a way to make the watch remember past device? And if not, is it planned?

Thanks !

thiswillbeyourgithub, 2022-09-26:
Up @daniel-thompson 

Do you know how it could be possible to access the list of connected device from the watch? I'm so sick of having to reconnect the watch to the phone after every new flashing that I end up never connecting it...

daniel-thompson, 2022-09-30:
I don't really understand the question. The watch's MAC address is set in the fuses during manufacturing, reflashing it shouldn't have any affect on GadgetBridge's ability to find it.

thiswillbeyourgithub, 2022-10-03:
I was convinced for some reason that my watch kept failing to connect to my phone after a refresh but I checked and it works fine. Closing this, sorry :)

closed 2022-10-03 by thiswillbeyourgithub