wasp-os/wasp-os #343: Inverted display colours & mirrored display.

Umi-L, 2022-07-09:
I just recently installed wasp-os on my p8 MOY-TON5 and have been having issues with the display. 
The display seems to have inverted colours and flipped.

To get the install working I followed the solution proposed by fanoush in this issue https://github.com/daniel-thompson/wasp-os/issues/222 however this issue persists even within the bootloader.

An image of the issue is shown below
[image]

daniel-thompson, 2022-07-11:
> Just an update, figured out the colour issue. I solved it by calling  self.invert(False)  in the st7789.py file at the end of init_display. However the mirroring still persists.

Good work so far. Take a look at the st7789 and look at the stuff that controls what axes get updated when we address the screen. I suspect I've of this can mirror the coordinates.

Umi-L, 2022-07-11:
Just a final note before I close the issue as I figured it out. I solved the mirroring issue via reading the [ST7789 datasheet](https://www.rhydolabz.com/documents/33/ST7789.pdf) and finding the correct hex value to send in line 60 of the driver file being 01000000 or hex 40.

`60.     (_MADCTL,   b'\x40'), # Left to right, top to bottom`

closed 2022-07-11 by Umi-L

tomchiverton, 2023-07-12:
@Umi-L did these changes get committed ? Can the two changes be made without rebuilding and reflashing ?

tomchiverton, 2023-07-12:
I'd hoped that wasptool could do this, but sadly $ ./tools/wasptool --exec /home/falken/workspace/wasp-os-Elenedeath/wasp/drivers/st7789.py --eval "watch.touch = ST7789(watch.i2c)"

[##################################################] 100% 
~~~
MemoryError: memory allocation failed, allocating 501 bytes
Traceback (most recent call last):
  File "<stdin>"
SyntaxError: invalid syntax
~~~
~~~
Traceback (most recent call last):
  File "<stdin>", in <module>
TypeError: function takes 3 positional arguments but 2 were given
~~~

`./tools/wasptool --eval "watch.display.invert(False)"` takes care of the first change required

`./tools/wasptool --eval "watch.display.write_cmd( 0x36 ); watch.display.write_data(b'\x40')"` takes care of the rest.

This is non permanent and will be lost at reboot I think

tomchiverton, 2023-07-15:
I've tried rebuilding Elenedeath's branch with this change, but get trapped in a bootloop (wasp OS logo appears for a few seconds, then reboots. No "start", arrow or Bluetooth icon. 

Managed to get it back in OTA mode (start, arrow, Bluetooth logo) by some combination of a 5 minute charge and long or short holds of the button.

I then re-uploaded their binary micropython.zip with dfu.py and it's back to working. Hmm.

IDK if this is my local compile or not though.

tomchiverton, 2023-07-15:
What does work is adding 

```
watch.display.invert(False);
watch.display.write_cmd( 0x36 ); watch.display.write_data(b'\x40');
```

before the `wasp.system.schedule()` in wasp/main.py and then update it with 

`tools/wasptool --upload wasp/main.py`

then restart 

`tools/wasptool --reset`

and the display is fine.