wasp-os/wasp-os #345: stuck because cannot delete user storage

thiswillbeyourgithub, 2022-07-20:
Hi!

While playing around with sleep logging for a while, I ended up saving a lot of files on my pinetime.

The issue is that my folder `logs/sleep` contains so many files that I get a memory error trying to display it, so I can't remove those files because I can't get their names.

I tried playing around with things like  `for l in os.listdir("logs/sleep"): os.remove(l)` but I still get the memory error, even by launching it in the first line of main.py.

Any idea how I could solve this? I stumbled upon `os.ilistdir` [link](https://docs.micropython.org/en/latest/library/os.html) which returns a generator instead of a full list but it does not seem implemented in waspos.

daniel-thompson, 2022-07-25:
Take a look at https://github.com/daniel-thompson/wasp-os/blob/master/wasp/boards/pinetime/watch.py.in#L107 .

I think the simplest way to reset this would be to backup anything you need from the filesystem and then boot into safe mode, umount /flash and run mkfs by hand.

daniel-thompson, 2022-07-25:
(or just build a custom wasp-os binary with lines 103-106 commented out and the except block outdented... then go back to a normal build)