wasp-os/wasp-os #346: Potential Ram Issue when Registering Custom Apps

ArcherThe3rd, 2022-07-20:
When utilizing wasptool to execute ./tools/wasptool --exec wasp/apps/myapp.py --eval 'wasp.system.register(MyApp())' I encounter an error that reads:
[############################......................] 56.9%     
~~~
FATAL: uncaught exception 20004680
MemoryError: memory allocation failed, allocating 1208 bytes
~~~
Given that I have no issue manually uploading the file via the wasptool --upload tool, I assume that this is not an issue with the actual memory of the PineTime watch, which I verified with the --memfree tool. Has anyone else experienced this issue when attempting to register their own custom applications on a PineTime watch and if so, what were your workarounds?

thiswillbeyourgithub, 2022-07-21:
Not sure but have you tried sending the app as binary?

I'm note sure you can use memfree to investigate this because (no experts at all on this) if the watch failed to load the app, it will clean what it tried to load.

ArcherThe3rd, 2022-07-21:
Understood, I followed the steps to sending the app as a binary file and it worked. Thank you

closed 2022-07-21 by ArcherThe3rd

tomchiverton, 2023-07-29:
This still happens. I get 

```
$ ./tools/wasptool --exec rogue.py --eval "wasp.system.unregister(RogueApp) ; wasp.system.register(RogueApp())"
Preparing to run rogue.py:
[#############################################.....] 91.2%     
~~~
FATAL: uncaught exception 20004630
MemoryError: memory allocation failed, allocating 992 bytes
~~~

Please reboot your device
```

When sending the minimal 51 lines of code to draw the "ticks" around an analogue clock face : https://codeberg.org/falken/wasp-os-apps/src/commit/ec6b794d6c2a2008ebc134b99781217051c3d0e6/rogue.py

Runs fine via "make sim", but build and uploading an .mpy binary and then launching via editing wasp/main.py, reuploading that and then rebooting the watch works fine

beardeddude, 2023-07-29:
This is expected behavior. These smart watches only have a few KB of RAM. Even a small py file is going to take a huge portion of that. Using a binary .mpy helps but it till has to be stored in RAM to be run. IME the only way to run apps of any real size is to freeze the data into the build. That is the only way that the code can be run directly from ROM.

The ability to run apps from apps from flash is cool but you would need that app to be tiny, the RAM to be very open and every time you reload your probably going to have less RAM as the little things build up.

Just my 2 cents