wasp-os/wasp-os #354: Resistor colour code face for wasp?

drtonyr, 2022-08-30:
Hackaday features all sorts of interesting things (that's how I got here) including a clock that shows the time using the resistor colour code (https://hackaday.com/2021/07/15/a-perfect-clock-for-any-hackers-ohm/).

This is trivial to write for WASP, so I've done it.  I'm very pleased with the first-pass result (https://photos.app.goo.gl/Sd7FKMj3tBuS8UXZ8).  Does the project want it?

kj7rrv, 2023-06-26:
Would you mind sharing the code for this?

drtonyr, 2023-06-26:
Not at all.  However, it's a while since I did any WASP work and things have changed.  Attached is the code.  I know that WASP is more configurable now, so it's easier to add code that doesn't get loaded by default.  Once I get up to speed again I have a few things to merge in.
[resistor_clock.py.txt](https://github.com/wasp-os/wasp-os/files/11871700/resistor_clock.py.txt)

drtonyr, 2023-10-12:
Closing this issue as now merged in #447

closed 2023-10-12 by drtonyr