wasp-os/wasp-os #355: Cannot switch back to SoftDevice 2.01 on P8b

gmgrasso70, 2022-09-02:
I have been playing around with Wasp-OS for a while and wanted to try other firmwares, such as ATCwatch, but following the procedure to switch back to SoftDevice 2.01, by flashing this file DS-D6-adafruit-back-to-desay-sd132v201.zip, does not work. I have tried Daflasher App and NRF connect App, as well as a Linux PC, but the flashing starts and then hangs. Any suggestions?

GaryM0101, 2022-09-02:
On a sealed device there is no really good way of going back. I had this experience with Espruino. You'll need to open the p8 and get to the SWD port and re-flash the Wasp bootloader.hex

gmgrasso70, 2022-09-05:
Hi GaryM0101, I can flash the wasp bootloader and the Micropython zip, I cannot flash only the DS-D6-adafruit-back-to-desay-sd132v201.zip for some reason...

GaryM0101, 2022-09-05:
One more thing to try first, in the nRF app connect settings try reducing the number of packets to receive before sending "PRN" value. This will slow down the upload time and give a chance for the P8 to make it's updates. I heard folks were able to overcome aborts this way but some others weren't so lucky.

gmgrasso70, 2022-09-13:
Tried and did not work at all :-(

thiswillbeyourgithub, 2022-09-13:
Please provide pictures of the steps you took, especially in nRF connect