wasp-os/wasp-os #357: instruction to add ulab (numpy equivalent for micropython)

thiswillbeyourgithub, 2022-09-07:
Hi!

I discovered recently [ulab](https://github.com/v923z/micropython-ulab) which apparently can bring lots of functionnality of numpy to micropython. I am interested in some of those functions to handle computation to optimize my sleep alarm project ([SleepTk](https://github.com/thiswillbeyourgithub/sleep_tracker_pinetime_wasp-os)).

However I don't know C nor the first bit about hardware and have really no understanding of how to "add ulab to waspos".

Can anyone point me to ressources on how to have a go at this?

Thanks a lot!

drtonyr, 2022-09-07:
If you follow the build instructions, what size is the result?  My guess is that we are really tight for space and that ulab is too big to fit - but that's only a guess (the page suggests that an RPi build is quite easy).

daniel-thompson, 2022-09-08:
In theory you can follow a varient of the these instructions https://github.com/v923z/micropython-ulab#stm-based-boards (e.g. put ulab into the USER_C_MODULES path).

Note that the wasp-os wrapper makefile automatically places the [wasp/modules](https://github.com/daniel-thompson/wasp-os/tree/master/wasp/modules) into this variable so in principle you can just put the softlink to ulab here and the micropython.mk file should be automatically discovered.

Be warned about the code size though. You'll probably have to disable a bunch of built in applications to make space for the ulab code.

thiswillbeyourgithub, 2022-09-08:
Thanks a lot for the quick response!

@drtonyr : 
> ulab implements close to a hundred functions and array methods. At the time of writing this manual (for version 4.0.0), the library adds approximately 120 kB of extra compiled code to the micropython (pyboard.v.1.17) firmware. However, if you are tight with flash space, you can easily shave tens of kB off the firmware. In fact, if only a small sub-set of functions are needed, you can get away with less than 10 kB of flash space. See the section on [customising ulab](https://micropython-ulab.readthedocs.io/en/latest/ulab-intro.html#Customising-the-firmware).

From https://micropython-ulab.readthedocs.io/en/latest/ulab-intro.html

@daniel-thompson Thanks a lot for the guidance.

I think I managed to do some things by adding a link from `wasp/modules/code` to `ulab/code` then `make BOARD=pinetime all`.

I see a bunch of lines like : 
`
CC /redacted/wasp-os/wasp/modules/code/numpy/fft/fft_tools.c
CC /redactedwasp-os/wasp/modules/code/numpy/filter.c
`
si I figured I'm on the right track?

However after flashing micropython.zip and asking the remote console `from ulab import numpy` I get `ImportError: no module named 'ulab'`

Is there some obvious things I'm doing wrong? Do I have to flash the bootloader or is micropython.zip enough?

drtonyr, 2022-09-22:
> 

> By the way, wouldn't it be worth it to try and use ulab's FFT function to measure the heart rate? The current version is too unreliable to be used while running etc IMO

I know quite a bit about signal processing (40 years worth including teaching it at uni level), if you want a reliable heart rate then I'd use [autocorrelation](https://en.wikipedia.org/wiki/Autocorrelation) not FFT in the first instance.  You can peak pick the autocorrelation function and have something that reacts smoothly and with low latency and reasonable efficiency.  FFT is complex (pun intended) and gives you more than we need as it gives all frequencies.  I've done a lot of this sort of stuff, if we really have a problem with the heart rate measurements then we should collect data as to when it works and when it doesn't.

drtonyr, 2022-09-22:
I'd disagree.  ulab is very heavy, it's a nice-to-have if we had a lot of space, but we have nowhere near the space needed.
If we want better HRS then it should be done independently of ulab, it's PPG.get_heart_rate() that needs to be improved.  It takes a sample, high pass filters to get rid of DC, automatic gain control to get rid of amplitude, then low pass filters (I don't know why, I wouldn't do this).  It then does _compare() a root-mean-square operation with a buffer, but it does it for every possible shift.  That's a lot of bytes to ints conversion and a lot of multiplications.   There would be a lot less to do if we did:

acf[n] = alpha * agc[n] + (1-alpha) * spl[t] * spl[t - n]   for all n
where spl[t - n] is a buffer of past values.

We may find that PTAGC() isn't needed either.

Anyway, this is getting off track.  My point is that we should separate ulab and heart rate.  If we want better heart rate there's lots of easy wins.   ulab is large, if we want it then we should cut a load of stuff out and have a core release and space for optional things that individuals want, like ulab or their own code.

thiswillbeyourgithub, 2022-09-22:
Thanks again for your insight.

I indeed tested without the LPF and the AGC and it seems to work fine. But I noticed also that `_compare` does not actually use its 'shift' argument.

I also struggle to understand what you mean by "acf" and "alpha" in your example. Would you mind explaining a bit more?

drtonyr, 2022-09-22:
> I indeed tested without the LPF and the AGC and it seems to work fine.

I don't know the background here.  Perhaps there is a reference implementation that was ported to micropython?   There is some interesting comments in the code.

> But I noticed also that `_compare` does not actually use its 'shift' argument. It is normal?

Good catch.  Compare uses the shift, by the time _compare sees the data it's already shifted.

> I also struggle to understand what you mean by "acf" and "alpha" in your example. Would you mind explaining a bit more?

acf is an exponentially smoothed [Autocorrelation function](https://en.wikipedia.org/wiki/Autocorrelation).  acf is a pretty horrible acronym, but it's standard.   Autocorrelation just means the correlation of a signal with itself.   My pseudocode keeps the values for all lags, the highest value is the largest correlation, the number we want (you can get fractional values from interpolation).

alpha is the exponential decay.   The idea is that we want to do smoothing as efficiently as possible, so we could just keep 48 samples at 24Hz, or 2s worth of data and pick the lag which has the highest value.  If alpha is low then the past is forgotten quickly and there is little smoothing.  If alpha is high then we get a nice smoothed output.

We don't need a heart rate more frequently than once a second and we could use this to get a lot more efficiencies and more accurate measurements. 

The way to get really good heart rate code is to enable raw logging and collect data (the streaming raw data at 24Hz).   You then make things update as fast as you can every second whilst getting a really low number of half/doubling errors.

thiswillbeyourgithub, 2022-09-22:
Thanks again.

I ended up trying quite a lot of stuff and finely gave up, I don't have your knowledge and was procrastinating anyway :)

But I ended up still improving the heart rate app and will make a PR shortly!

closed 2022-09-22 by thiswillbeyourgithub