wasp-os/wasp-os #358: Display frozen at home screen

lenikhilsingh, 2022-09-12:
Hi,

My Pinetime watch after flashing wasp os is frozen on home screen:

[image]

Now, I want to go back to Infinitime. I'm not able to do it.

The DFU button is not appearing when I try DFU with NRFConnect:

[image]

As for Gadetbridge, the device gets disconnected and is not able to connect when I reboot the watch by holding the button until the pine cone arrow appears. I tried another way, by click opening the firmware file with gadgetbridge. Here's the response:

[image]

I tried DaFlasher as well. DaFlasher's response:

[image]

Please suggest ways to rectify this. So that I can reinstall Infinitime.

thiswillbeyourgithub, 2022-09-12:
What version of nRF connect are you using?

Maybe take a look at this https://github.com/daniel-thompson/wasp-os/issues/335#issuecomment-1170212449

lenikhilsingh, 2022-09-12:
Hi, thank you so much for replying.

I'm using [nRF Connect 4.24.3](https://github.com/NordicSemiconductor/Android-nRF-Connect/releases/tag/v4.24.3)

lenikhilsingh, 2022-09-12:
I'm not able to use dfu, because DFU button doesn't appear in my case. As opposed to https://github.com/daniel-thompson/wasp-os/issues/335#issuecomment-1170212449

thiswillbeyourgithub, 2022-09-12:
What is the name of the pinetime device you're trying to connect to, as seen by nRF connect ?

Is it something like "pinetime" or "pineDFU" ?

lenikhilsingh, 2022-09-12:
Name is PINETIME. Anyway, I was wondering, why does it show CONNECTED NOT BONDED. I might not have noticed earlier.

thiswillbeyourgithub, 2022-09-12:
You have to press the button until the watch reboots into the (IIRC) bootloader mode. In bootloader mode, pressing again the button will boot wasp os.

The DFU button will only appear in bootloader mode.

lenikhilsingh, 2022-09-12:
yes, I did reboot by hard pressing the power button for sometime. It was on pinetree logo(and start arrow)

The device got disconnected from NRF connect. And as long as it remains on the pinetree logo, I'm not able to connect with the device on nrf Connect. When I press the power button to come back to home screen, then the device gets connected to nrf connect.
But then, after connecting, I'm not able to access DFU button.

[image]

[image]

lenikhilsingh, 2022-09-12:
Is there any simpler process to reset the watch and go back to the initial OS, that it came installed with?

thiswillbeyourgithub, 2022-09-12:
I think you're misuing nRF connect. When switching from "regular mode" to "bootloader" the device changes name from "pinetime" to "pineDFU".

Your picture is showing you're on the wrong device in nRF connect

lenikhilsingh, 2022-09-12:
It's the same device. Please check the device ID, F4:27:4C:72:75:F8. When I switch from regular to bootloader mode I'm not able to connect. Anyway, what do you think might have gone wrong, when you say I'm misusing the app?

thiswillbeyourgithub, 2022-09-12:
the tab name in nRF connect is "pinetime" and I think it sould be "pineDFU". Search again for the device once you're in bootloader mode.

lenikhilsingh, 2022-09-12:
Actually, This is an important piece of information that I missed. I wasn't expecting the name to change to PineDFU, that's why I wasn't scanning for the device again. So, When I scanned again, the device  name "PineDFU" appeared and I flashed the reloader zip: reloader-infinitime-recovery-0.14.1.zip.

Anyway, After this when the flashing was complete, the watch has gone in the loop, where in these two screens appear one after another.

[image]

[image]

I'm checking @daniel-thompson's comment https://github.com/daniel-thompson/wasp-os/issues/335#issuecomment-1170212449, maybe I can relate with that issue.

lenikhilsingh, 2022-09-12:
Okay, I followed these steps And flashed pinetime-mcuboot-app-dfu-1.9.0.zip

[image]

Now, the device is frozen at Infinitime's home screen:

[image]

lenikhilsingh, 2022-09-12:
Thank you. We cna close this issue now.