wasp-os/wasp-os #359: Heart Rate Monitor debug mode does not work as advertised in the documentation

jcp-sd, 2022-09-13:
I tried to download some raw data from the PPG sensor using the instructions given in the Application Library -> Builtin -> Heart Rate Monitor section of the manual that I briefly summarize here;

```
./wasptool --eval 'wasp.system.app.debug = True'     # to enable

./wasptool --pull hrs.data     # to download the collected data

```

I tried this, but no data was collected (the pull failed with an error, whereas I could pull other files with no problems). After examining the source code, it appears this is not surprising. The PPG class has code to save the data and write it to the file, but it _**doesn't ever get enabled**_. The variable wasp.system.app.debug is never mentioned and the debug() method of the HeartApp is never called (that was probably supposed to happen in register method of the Manager class where the specific features of the default apps are handled).

Judging from the documentation, I am guessing this code just didn't get written or it got eliminated by an overlapping change from someone's pull request. I did not put together a fix and pull request under the assumption that the original author might have had something else in mind.

If you got here via a web search because you are having this same problem, here is a work around that I successfully used (as of this writing) to generate some raw PPG data and download it;

```
./wasptool --device 00:DE:AD:BE:EF:00 --eval "wasp.system.quick_ring[3].debug = True"     # to enable

./wasptool --device 00:DE:AD:BE:EF:00 --eval "wasp.system.quick_ring[3].debug = None"     # to disable

./wasptool --device 00:DE:AD:BE:EF:00 --pull hrs.data     # to download the results

./wasptool --device 00:DE:AD:BE:EF:00 --console     # to remove old data file
>>>
>>> import os
>>> os.remove('hrs.data')
Ctrl-X
```

daniel-thompson, 2022-09-25:
> I tried to download some raw data from the PPG sensor using the instructions given in the Application Library -> Builtin -> Heart Rate Monitor section of the manual that I briefly summarize here; [...]

These instructions miss out the "select the heart rate application using the watch UI" part! If you did not switch to the HR app before running the first of the commands above then the debug flag will be set in the wrong application (which probably ignores it). Having said that, the instructions could perhaps be clearer about the need to restart the HR app in order to activate the capture code after setting the debug flag.

Note that the reason the documented instructions do not use quick_ring to set the debug flag is because some users may have installed additional applications into the quick ring.

jcp-sd, 2022-09-29:
Thanks Daniel, that did work for me.

Roger your comment about the quick_ring variable. I knew that was an ugly hack since the index would not always be correct.

I am still a little mystified why this works. The wasp.system.app variable appears to be the currently loaded app, but that was not at all obvious to me after all my examination of the code. I've been doing Python since the 90's but not much in the last few years, so I am a little "rough around the edges" there.

Anyhow, an additional note to start the HRM first in the instructions in the documentation (and also in the comments in heart.py) would be in order given that there is a push to collect raw HRS data in different settings.

daniel-thompson, 2022-09-30:
> I am still a little mystified why this works. The wasp.system.app variable appears to be the currently loaded app, but that was not at all obvious to me after all my examination of the code. I've [...]

AFAIK the only bit of "Python" that is happening here is the knowledge that pretty much every data structure in Python is copied by reference (rather than value).

```python
stingray$ python3
Python 3.10.7 (main, Sep  7 2022, 00:00:00) [GCC 12.2.1 20220819 (Red Hat 12.2.1-1)] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> a = {}
>>> b = a
>>> b[10] = 'This will appear in a'
>>> print(a)
{10: 'This will appear in a'}
>>> 
```

The above example uses a dictionary but objects namespaces work the same way. After that it's simply a matter of updating the reference to the current app during a switch:
https://github.com/daniel-thompson/wasp-os/blob/master/wasp/wasp.py#L269

For as long as the HR app is in the foreground then wasp.system.app and wasp.system.quick_ring[3] are synonyms.

jcp-sd, 2022-10-02:
Thanks Daniel.

I was indeed aware of that aspect of Python. It's something you need to understand or _many things_ are going to look rather _"mysterious"_.

Anyhow, having two different names for the same object can reduce to just one line of code that may not jump out. I have always put conspicuous comments to that effect in my Python code so that it is obvious to the more casual observer.