wasp-os/wasp-os #364: OSError: Can not notify attribute value. status: 0x08

drtonyr, 2022-09-23:
I'm writing a soft keyboard (which is quite fun!) and I've suddenly started getting:

`OSError: Can not notify attribute value. status: 0x08`

grep tells me that this line comes from micropython/ports/nrf/drivers/bluetooth/ble_drv.c which is very strange, my code is executing a standard `def touch(self, event):` and I'm doing nothing more complex than putting characters on the screen and working out which bounding box the touch event came from.   The (unfinished) code is below, the message is triggered by wasp.watch.drawable.set_font(fonts.sans24), comment it out and things work.

I just wondered if anyone else had seen this (a search of the archives suggests not).  Maybe it's just an out-of-memory error that causes some random corruption and crash?

Tony

P.S. I know this code is messy and broken, but it still shouldn't throw the error above.

```
  def touch(self, event):
    if self.mode:
      y = event[2] // 80
      if len(self.mode[y]) == 3:
        x = event[1] // 80
      else:
        x = event[1] // 120

      self.text += self.mode[y][x]
      self._draw()
      self.mode = None

    elif event[2] >= 120:
      y = event[2] // 24 - 5
      if y != 2:
        x = event[1] // 24
      else:
        x = max(event[1] - 12, 0) // 24
        x = min(x, len(layout[y]) - 1)

      wwdraw.fill()
      x = min(max(x, 1), len(layout[y]))
      y = min(max(y, 1), 3)

      self.mode = [layout[y-1][x-1:x+2], layout[y][x-1:x+2], layout[y+1][x-1:x+2]]
      wasp.watch.drawable.set_font(fonts.sans24)
      for i in range(3):
	nchar = len(self.mode[i])
	offset = 0 if nchar == 3 else 40
	for j in range(nchar):
          wwdraw.string(self.mode[i][j], 80 * j + offset, 80 * i + 20, width=60)
```

thiswillbeyourgithub, 2022-09-23:
Can this be caused by not using `draw = wasp.watch.drawable ; draw.set_font(fonts.sans24` ? Pretty sure it should not change anything but worth a try.

All `set_font` is doing is storing the argument in the instance of Draw565, maybe there is something funky going on because the class gets garbage collected?

Looking at your code I see that wwdraw is not declared in your example which might be important.

Actually why don't you call set_font from wwdraw?

drtonyr, 2022-09-24:
> Looking at your code I see that wwdraw is not declared in your example which might be important.
> 
> Actually why don't you call set_font from wwdraw?

Thank you very much @thiswillbeyourgithub - that was the problem.

I have `wwdraw = wasp.watch.drawable` as a global variable, so, as far as I my understanding went, it really didn't matter if I called via `wwdraw` or `wasp.watch.drawable`, they were the same thing.  However, it did matter, calling everything from `wwdraw` (which others call `draw`) fixed the problem.  I clearly don't know python/micropython/wasp as well as I thought I did.

Thanks again!

closed 2022-09-24 by drtonyr