wasp-os/wasp-os #365: DATA_SIZE_EXCEEDS_LIMITS when using wasp tool for flash CI build on P8

abishekmuthian, 2022-09-23:
Greetings Daniel and other contributors.

I tried to flash couple of latest CI builds on P8 using wasp tool and I'm faced with the following error - 

`Exception at line 166: Error in START_DFU procedure, reason: DATA_SIZE_EXCEEDS_LIMITS`

Size of `micropython.zip` in release build (0.4.1) is `293.1 KB` but on CI build it is `327.1 KB`, Release build flashes successfully.

drtonyr, 2022-09-23:
What does `arm-none-eabi-gcc --version` say?   I remember when I started I used an 'easy' way to install gcc for arm (apt install) which, IIRC, gave version 9 and the result was too big.    https://github.com/daniel-thompson/wasp-os/issues/294 was my thread and see also https://github.com/daniel-thompson/wasp-os/issues/219.

abishekmuthian, 2022-09-24:
Hi, The issue I pointed out is not when building but when flashing the pre-built binary from CI to the P8 via wasp-tool, The error mentioned is printed by the wasp-tool.

drtonyr, 2022-09-24:
Humm, strange, it works for me.

````
unzip p8-752fe77376b3314e69cf2989b7010cd05f8ee7de.zip 
Archive:  p8-752fe77376b3314e69cf2989b7010cd05f8ee7de.zip
  inflating: bootloader-daflasher.zip  
  inflating: bootloader.hex          
  inflating: micropython.zip         
  inflating: reloader.zip            
think0 tonyr: tools/ota-dfu/dfu.py -z micropython.zip -a F2:61:81:68:9B:F3 --legacy
Sending file firmware.bin to F2:61:81:68:9B:F3
Binary imge size: 334260
Binary CRC32: 655586686
Connecting to F2:61:81:68:9B:F3
Waiting for Image Size notification
Waiting for INIT DFU notification
Begin DFU
[##################################################] 100.0%  (326 of 326 kb)

Upload complete in 6 minutes and 31 seconds
Waiting for DFU complete notification
Waiting for Firmware Validation notification
Activate and reset
```
What do you get when you do the same as I did?

abishekmuthian, 2022-09-25:
Thanks for trying it out, I did the same before and I tried again with same results -

```
unzip p8-752fe77376b3314e69cf2989b7010cd05f8ee7de.zip 
Archive:  p8-752fe77376b3314e69cf2989b7010cd05f8ee7de.zip
  inflating: bootloader-daflasher.zip  
  inflating: bootloader.hex          
  inflating: micropython.zip         
  inflating: reloader.zip

./dfu.py -z ~/Downloads/micropython.zip -a XX:XX:XX:XX:XX:XX --legacy
Sending file firmware.bin to XX:XX:XX:XX:XX:XX
Binary imge size: 334260
Binary CRC32: 655586686
Connecting to XX:XX:XX:XX:XX:XX
Waiting for Image Size notification
Exception at line 166: Error in START_DFU procedure, reason: DATA_SIZE_EXCEEDS_LIMITS
```

Additional information which might be useful -

1. The device is P8a i.e. the one with SC7A20 accelerometer.

2. Before flashing the system (micropython.zip) from the CI build I tried to flash the boot loader from the same, It failed in the DaFlasher, I tried reloader.zip in the NRF connect and it was struck at 100%, So I aborted it. Then I tried to flash the boot loader from the release build and it failed in both Da Flasher + NRF Connect. Flashing micropython.zip from release worked in Da Flasher. Could I have messed up partition with my bootloader flashing shenanigans? (Note that I can flash micropython.zip from release and boot the system fine even after all these).

3. Flashing CI micropython.zip using Da Flasher also results in `DATA_SIZE_EXCEEDS_LIMITS` just like with wasptool.

daniel-thompson, 2022-09-30:
Seems almost like the bootloader your using doesn't have this patch (which increases the maximum image size that can be written): https://github.com/daniel-thompson/wasp-bootloader/commit/1d128f1311c084c23a662cb68263df9b187ebf94 .

It's odd that you can't update it with the reloader though.

abishekmuthian, 2022-10-01:
Thanks for the insight Daniel, I wouldn't have been able to find that.

Is reloader.zip the expected way for someone to update their bootloader? I found only discussions on PineTime to upgrade from infinitime reg reloader.zip  and so I thought I made a mistake by flashing it on P8.

Perhaps flashing an older reloader.zip, Say like from v0.3 release might apply that patch for me? I'll try that and let you know.

fgaz, 2023-01-15:
@abishekmuthian did that work in the end?

abishekmuthian, 2023-01-15:
@fgaz I couldn't try it again, I will do it soon. Is it happening for you as well? 

Did you try any other method to update your bootloader?