wasp-os/wasp-os #367: P22(d) support

j-mizzle, 2022-09-26:
Is there any chance of support being added for the P22 devices? I would live to see Wasp-OS or Infinitime on this like I'm able to do with my Pinetime but I have no knowledge on the subject. It is supported by eucWatch, so I assume it would be open enough to run other firmwares. Using their Pinetime/P8/P22(*) branch works but selecting the actual P22 branch boots with a mirrored display. I assume it shares a lot of hardware with the Pinetime and P8 with maybe a different pin layout. I'm personally not concerned with the heart rate sensor as I don't use that even on the Pinetime, just a watch with notifications.

daniel-thompson, 2022-09-30:
If you know the pinout I can help you find the bits of the bootloader and wasp-os that you would need to update.

j-mizzle, 2022-10-15:
> 

I wouldn't know where to begin or how to find that information. All I know is it seems to share a lot in common and be very similar to the P8 and PineTime. Like I said eucWatch firmware runs and seems to reuse a lot of P8 & PineTime bits, so hopefully somebody more knowledgeable than I can help.

j-mizzle, 2022-11-30:
> If you know the pinout I can help you find the bits of the bootloader and wasp-os that you would need to update.

Could you explain how I can find the pinout? I've got 2 p22d's and am willing to test things

jawsh207, 2024-01-22:
> If you know the pinout I can help you find the bits of the bootloader and wasp-os that you would need to update.

I'm not sure what the pinout is or how to find that information.

drtonyr, 2024-09-07:
So I found a P22 on ebay for not-much and thought I'd see how far I could go before bricking it.

I followed the P8 workflow at https://wasp-os.readthedocs.io/en/latest/install.html#installing-wasp-bootloader and used DaFlasher to install DaFitBootloader23Hacked.bin and FitBootloaderDFU2.0.1.zip.  I was quite pleased that this worked, the screen was flipped and the colour map was inverted, but it worked.

I then ran `tools/ota-dfu/dfu.py -z build-p8/bootloader-daflasher.zip -a D8:81:7F:DE:1F:58  --legacy`.   This reported `Waiting for Firmware Validation notification` and `Activate and reset` as it should, but the watch was then bricked.

I may find another P22 on ebay one day.