wasp-os/wasp-os #371: P8 plus, P8 plus GT and P8 Pro compatibility?

LinyeNinja, 2022-10-04:
In my country I can only find the P8 Plus, Plus GT and Pro on sale, and looking at the Colmi website it seems that they are all the same watch with different software, battery sizes and resolutions (although the Pro is not listed). 
Is this the case? Has someone tested the firmware on the Plus watch or knows if it shares the same chip?

Thanks in advance.

leroyshirto, 2022-10-31:
It appears the P8 Plus uses a different micro controller. https://github.com/daniel-thompson/wasp-os/issues/259

I have raised a PR to reflect this in the docs https://github.com/daniel-thompson/wasp-os/pull/376

leroyshirto, 2022-10-31:
As a side note. Prior to realising it was a different controller. I tried flashing the P8 Plus with the DaFlasher Android app.

The app detected a Da-fit device. When I tried loading ` DaFitBootloader23Hacked.bin`. The watch switched to a screen saying "Upgrading" but the DaFlasher app crashed and returned to the list of BT devices

nicholas-gh, 2023-05-01:
The 'P8 Max' is also GR5515 based according to https://www.colmi.com/p-series-smartwatch/

80avin, 2025-01-01:
Adding for reference, there is a micropython implementation for GR5515 mcu at https://github.com/goodix-ble/GR551x-MicroPython

Likely, the files in this micropython port can be used as a stepping stone.

I am eager to do this but I have almost zero experience in embedded space and neither I find much time out of day job.
If someone familiar with the codebase can give some direction, I can work on porting wasp-os to GR5515.

mbeijen, 2026-11-14:
Would be great to support any Colmi, at the moment I don't have access to this hardware. It would require a dedicated person maintaining this platform, if they step up, it sounds like a good plan to add support for this

drtonyr, 2026-11-14:
> Would be great to support any Colmi, at the moment I don't have access to this hardware. It would require a dedicated person maintaining this platform, if they step up, it sounds like a good plan to add support for this

Would you have time if you had the hardware?  I have a single P8 that works, although I haven't played with it for a long time.  I also have several surplus P8 variants that have the magnetic charger. IIRC the P8B.  If you really had time I'd be happy to post you my one good working P8 and one of the newer versions.

mbeijen, 2026-11-15:
> Would you have time if you had the hardware? I have a single P8 that works, although I haven't played with it for a long time. I also have several surplus P8 variants that have the magnetic charger. [...]

Well, as stated in this issue, the P8 Plus/Pro is a _completely different_ watch than the P8. And while there is MicroPython support for the chipset, all peripherals probably need work still. Also, Colmi (as opposed to Pine64) does not have hackability really top of mind afaik.

So while the P8 at least _used_ to work on PineTime the P8 Plus definitely does not. And then I'd prefer to focus on the PineTime and the upcoming PineTime Pro.

I might be interested in the P8 as it would allow me to really test PineTime on the platform. For the P8 Plus/Pro, I would be open to do PR review if someone would step up as a platform maintainer, but I'm not interested in being that person myself ;-)

I hope you understand!

drtonyr, 2026-11-15:
Sorry, I was both hijacking a thread and I wasn't clear enough - two errors in one post.   This is a P8 plus, P8 plus GT and P8 Pro compatibility thread, I'm not talking about any of those, I'm talking about P8 and P8b, two watches that already have support.

If you or anyone else would like a P8 (mechanical charger clip) or P8b (magnetic charger) - then do let me know.   Neither of these say Colmi P8 on them - they are just cheap eBay purchases, but the P8 does boot into WaspOS and the P8b does say "P8b MOY-TON5-2.0.1".   This is just to help you develop and support what is already supported, no request to port to something new.

No worries if not - and no need to reply.  I'll just apologise again for hijacking the thread - sorry!

mbeijen, 2026-11-15:
No worries and thanks for clearing this up. Sending me a P8 would be great, thanks for the offer; I sent you an e-mail (scraped your email from your github commits) with my address.