wasp-os/wasp-os #380: [feature request] Improvements to heart rate application

jcp-sd, 2022-12-08:
This is a "catch all" issue for the discussion of improvements to the existing heart rate application.

The current "wish list" includes, (but is not limited to) the following items;

1. Fix the known problems with the polling period of HRS samples as a function of the bit resolution. This topic was discussed and mostly clarified in pull request #363.
2. Fix the code that polls the HRS3300 so that higher HRS sample rates can be used without dropped or repeated samples.
3. If possible, determine how the ambient light sensor of the HRS3300 interacts with the HRS sampling (Is correction for ambient light automatic, or does it require explicit post-processing of ALS samples? The data sheet provides no solid guidance on this.)
4. Modify the heart rate estimation algorithm to provide more robustness in the presence of noise that is a result of motion.

jcp-sd, 2022-12-11:
I will kick this off by making my rewrite of the heart rate app available. I didn't have a finished product in mind when I wrote this. It's more of a platform for further development, and a proof of concept of sorts (it mostly "keeps up" with 48 millisecond polling, but still drops samples every so often).

It is on a separate branch "hrm-development" at this repo; https://github.com/jcp-sd/wasp-os/tree/hrm-development

It does fix issue 1 above. It provides an improved scheme for polling the HRS3300 at more precise intervals. It includes a new IIR filter class that supports filters with an arbitrary number of poles and zeros.

The heart rate estimation algorithms operate on a buffer of filtered samples of configurable length (10 seconds is the default). Two algorithms are presently implemented; one based on a peak finder, and the other based on the sample autocorrelation function. To avoid dropping PPG samples from one massive computation when the buffer is full, much of the computational effort has been "spread out" over subticks, and performed after the PPG sample has been filtered.

When I hold still, it agrees pretty well with my MI Band 5 on the other arm. The bad news is that it is not very robust against motion at this point. (It currently doesn't use any accelerometer data).

Here is a picture of the display;

[image]

thiswillbeyourgithub, 2023-01-16:
Hi!

Any update on this feature by any chance? I would love to track my heart rate regularly especially during sleep but I'm wary of the accuracy given all the discussion in the other thread.

Is there a clear roadmap of things to do? I could maybe help in some ways.

Thanks.

drtonyr, 2023-01-16:
I was very active in the heart rate discussions and then I went silent - sorry about that!   I found that my cheap P8 was getting a very poor raw signal compared with what people were posting here (Pinetime).   I still like the signal processing work that I was doing, but I was fighting a battle that others didn't have.  Of course I'm generalising from very small samples, it may be that some Pinetime also give a lousy signal and maybe some P8 give a good signal, I don't know.

jcp-sd, 2023-01-17:
My work on this got sidelined due to the holiday and all that goes with that. The good news is that I had plenty of time to test my new code above in a variety of situations. As a result, I have identified a few additional changes that I hope to get implemented shortly.

I think the question of what the sampling period should be is settled. It _is_ a function of the setting of HRS_RES field in register 0x16. The table that @drtonyr posted in https://github.com/wasp-os/wasp-os/pull/363#issuecomment-1279733038 worked well for me, (reproduced again below).

| HRS_RES | Resolution | Period |
| ------- | -------------- | -------- |
| 0100 | 12 bits | 6ms |
| 0101 | 13 bits | 12ms |
| 0110: | 14 bits | 24ms |
| 0111 | 15 bits | 48ms |
| 1000 | 16 bits |  96ms |

One thing I learned while writing the new implementation was that updating the display is _very_ slow. When polling the HRS3300 at 48 milliseconds, it's only enough time to update a small portion of the above display or the next sample will get "dropped".

There are still a lot of unanswered questions related to the HRS3300 itself, and the algorithms to process the data (filters, etc.) as well. Getting feedback from others on how my changes work for them would help advance this effort.

I made sure to judiciously comment my work, so at some point it could be pulled into the main repo. It's a long way from being stable, so making it a development branch would make the most sense.

GaryM0101, 2023-03-17:
I cloned the hrm-delevopment branch but when I run the simulator I get:

File "/home/pi/jcp/wasp-os/wasp/ppg.py", line 40, in <module>
    _LEN_DEBUG_BUFFER = const(256)	# length of the debug, raw data buffer (samples)
NameError: name 'const' is not defined

Sounds like I need to declare it before we define the buffer, any suggestions?

The goal is to test this branch with the SleepTK app and compare the heart rate sleep cycle with the current rate we are using.

thiswillbeyourgithub, 2023-03-18:
Isn't it just a case of 'from micropython import const' ?

GaryM0101, 2023-03-18:
Yes and thanks! That's what I tried but didn't catch that I also had to define it in the heart.py as well. Now it works!

GaryM0101, 2023-03-18:
Tried this on the watch but it's using too much ram and I get the out of memory message, though checking the memory via wasptool didn't indicate I was low on memory. I suspect the heart app is trying to set the sample rate and bombing, the hrs light does activate before it bombs.

jcp-sd, 2023-03-18:
It's great to see some more interest in this !!!

About a month ago, I made some changes to the filter coefficients (in ppg.py) and got it working much better for me. I had some problems with my charging cradle that I eventually fixed by prying the back off and re-seating the little PC board inside. That, and increased work commitments kept me from doing any serious work on this until just recently.

As you discovered on your own, const() is specific to micropython. @GaryM0101- let me know how this works under the simulator. I never tried the simulator because I saw somewhere the HRS3300 is not emulated there. I wrote a very abbreviated equivalent of heart.py to call ppg.py with raw data I had collected to test things in a desktop python environment.

With regard to the memory problem, I'm struggling with that one myself right now. It's rather strange because it has happened on and off during my dev efforts after making seemingly small, and inconsequential changes to the code.

I made some small changes to wasp.py to explicitly call gc.collect() after a swipe that changes the active app. This helped at first, but then the problem re-appeared again later. That one liner change might be worth merging into the main branch.

I'm planning on running the code under desktop python together with a memory profiling module to see if there is an undetected leak somewhere. @GaryM0101 - you could try just reducing the size of the hrm sample buffer (see _LEN_HRM_BUFFER_SEC in ppg.py) from 10 sec to 8 or 9 and see if that helps.

The memory issue is really weird, and I ran across a post that seems to describe something quite similar. The responder mentions a known problem with micropython itself, but claims it was fixed. I'm planning to dig into the history of that, or contacting the person who stated that to get more details. See the post

https://forum.micropython.org/viewtopic.php?t=3252

My general feeling is that this could be made to work quite well under your SleepTK since the display update (rate) issues are not there.

GaryM0101, 2023-03-19:
Hi John, thanks for the detailed explanation and tips! I'll take a look and see what I can do to help. The simulator works great, dummy data is looped in from the watch.py in the boards/simulator directory. I would call it HRS3300 simulator more than a emulator. 
Cheers! Gary

jcp-sd, 2023-03-29:
@GaryM0101 Thanks much for sharing those results. Independent data points are very much appreciated.

I have been quiet, but looking into the memory issue quite closely. I'm convinced that it's a memory **fragmentation** issue. There is enough free memory available, but the largest available block isn't big enough (the sample buffer is pretty beefy for such a device). This somewhat explains the "spooky" and non-deterministic behavior I was seeing. (I was getting really-really worried that I had just done something incredibly stupid !!)

I have been testing some different mods to see how robust they are, and will post again when I have pushed them to my git repo.

I have not tested the raw data capability yet, so I will look into that soon. Getting more raw data out is definitely on the TO DO list. (Especially under conditions where the HR values appear to be wrong!) I don't have much confidence in the raw data I collected with the original heart.py code. Even after fixing the sample rate issues,  I suspect samples were getting dropped.

BTW, my code implements a new polling scheme that does fairly precise timing. It also checks for dropped samples and displays a running total. In your pic, "7" represents the total number of dropped samples.

Also, there is a problem with wasptool when you try to download a file from the watch unless it's really-really small. I think the python open()  on the watch side needs to specify an explicit buffer size (a small buffer size to avoid running out of memory). Or perhaps disable file buffering completely.

Much more to do...

thiswillbeyourgithub, 2023-06-24:
I don't know if this can help but here's a recent PR to infinitime about heart rate : https://github.com/InfiniTimeOrg/InfiniTime/pull/1486#issuecomment-1353695831

GaryM0101, 2023-06-24:
> I don't know if this can help but here's a recent PR to infinitime about heart rate : InfiniTimeOrg/InfiniTime#1486 (comment) [1]

I haven't tried the new infinitime release but will load it this
weekend. John aka jcp-sd should have a look at this and get his opinion.
Unfortunately it has no long term logging to to see how it performs over
time unlike how you can do with sleeptk. 

BTW way off topic... could sleeptk have the ability to track heart rate
only? This would be good for long term logging during the daylight
hours.   

Gary 

 

Links:
------
[1]
https://github.com/InfiniTimeOrg/InfiniTime/pull/1486#issuecomment-1353695831
[2]
https://github.com/wasp-os/wasp-os/issues/380#issuecomment-1605668362
[3]
https://github.com/

thiswillbeyourgithub, 2023-06-24:
Well i have a code somewhere to log the heart rate every x minutes during normal wasp operation. It's implemented as a new method to Manager class that is hooked to the settings app. If you are interested open an issue in the sleeptk repo and i'll take a look for you.

It's actually when I coded this that i noticed that walking or moving my hand would give me ventricular fibrillations OR that the measurement was off (hint: i'm mostly alive). One thing led to another and that's why this issue got created fyi.

jcp-sd, 2023-06-29:
> I don't know if this can help but here's a recent PR to infinitime about heart rate : https://github.com/InfiniTimeOrg/InfiniTime/pull/1486#issuecomment-1353695831

Thanks @thiswillbeyourgithub , that's an interesting and promising effort. I have been pretty busy with some other things (that I'm getting paid for!), but I will have a closer look when I get more time. I'm especially interested in what they are doing with the ALS (the ambient light sensor) as I don't have even a clue about that right now.

As far as implementing that in wasp-os, I'm skeptical that it can be done. The memory fragmentation issues I mentioned above in (https://github.com/wasp-os/wasp-os/issues/380#issuecomment-1487776744) is just a kiss of death and seemingly impossible to deal with. I usually optimistic, but my experience implementing my pretty simplistic algorithm has not been encouraging. I have two (seemingly) identical PineTimes with my own wasp-os images flashed to them. One can successfully allocate a 10sec buffer for the samples, the other can't!!!   WTF??????? The Infinitime algo would require another buffer of the same size, and FFT code that will gobble even more precious space.

For the Infinitime folks, they may want to look at some even better beat detectors. A good survey of open source beat detectors can be found in this article; https://iopscience.iop.org/article/10.1088/1361-6579/ac826d

Most of the algorithms mentioned in the article are targeted at offline processing on a workstation, so they can be very demanding of resources. Some of them might be ported to something like the Infinitime platform with some work, but many of them will be just too resource hungry for a smart watch. The source for many of those can probably be found somewhere on https://physionet.org/ (Try a google search formatted like this: "qppg site:physionet.org"  )