wasp-os/wasp-os #390: memory leak in simulator?

tigercoding56, 2023-01-16:
[image]

i was trying to develop a simple flappy bird clone using the simulator
[flappybird.py](https://github.com/wasp-os/wasp-os/files/10429865/flappybird.txt)
and  the emulator gave me a out of memory (only 25% of my computers memory was full )  
now the documentation states that "note the simulator does not have the memory restrictions of the hardware " (https://wasp-os.readthedocs.io/en/latest/appguide.html#testing-on-the-simulator)   and i am sure the test program i made should not consume more than 2.8kb of memory (its filesize * 2)      , it worked fine before trying to use lines to draw the pipes ( i used rectangles before)

fgaz, 2023-01-17:
Minimal reproducer:

```py
import wasp

class Bug390App():
    NAME = "Bug390"

    def foreground(self):
        wasp.watch.drawable.line(0,0,0,0,0xaaaaaa)
```

The problem is the color. `0xaaaaaa` is not a RGB565 color. You have to use something like `0xaaaa` (take a look at how other apps do it).

It shouldn't be reported as an OOM error though, that's for sure. The weird thing is that `fill()` reports a different error: `ValueError: memoryview: invalid value for format 'H'`.

thiswillbeyourgithub, 2023-01-17:
I was able to get the following error that might be more informative 👍 
```
Traceback (most recent call last):
  File "/REDACTED/wasp-os/wasp/wasp.py", line 565, in run
    self._tick()
  File "/REDACTED/wasp-os/wasp/wasp.py", line 528, in _tick
    self._handle_touch(event)
  File "/REDACTED/wasp-os/wasp/wasp.py", line 487, in _handle_touch
    self.app.touch(event)
  File "/REDACTED/wasp-os/wasp/apps/launcher.py", line 62, in touch
    wasp.system.switch(app)
  File "/REDACTED/wasp-os/wasp/wasp.py", line 306, in switch
    app.foreground()
  File "/REDACTED/wasp-os/wasp/apps/SleepTk.py", line 164, in foreground
    self.was_inited = self._actual_init()
  File "/REDACTED/wasp-os/wasp/apps/SleepTk.py", line 110, in _actual_init
    wasp.watch.drawable.line(0,0,0,0,0xaaaaaaaa)
  File "/REDACTED/wasp-os/wasp/draw565.py", line 411, in line
    px = bytes(((color >> 8) & 0xFF, color & 0xFF)) * (width * width)
OverflowError: repeated bytes are too long

```

daniel-thompson, 2023-01-18:
~~~python
wasp.watch.drawable.line(0,0,0,0,0xaaaaaa)
~~~

Does not draw a line whose colour is 0xaaaaaa! The fifth argument is the line width! The command will draw a line whose width is 11184810 pixels wide. This causes the line drawing code to allocate a 125099974736100 byte buffer to hold the square that is the minumum drawable for such a wide line.

In short there error has been reported correctly... even my most powerful machine can't allocate that much memory (which it is a little over 10 terabytes ;-) ).

PS I think it is also correct that a 32-bit machine would consider this an overflow error... since the memory requested does not fit into a 32-bit value.

fgaz, 2023-01-18:
Duh. I mixed up the arguments :facepalm: 

I'll restore the original title

fgaz, 2023-08-25:
Looks like this is solved, so I'm closing

closed 2023-08-25 by fgaz