wasp-os/wasp-os #391: pixel buffer

tigercoding56, 2023-01-18:
i am trying to develope a flappy bird clone for the pinetime smartwatch , and while it runs decently good for  smartwatch , i would like to be able to draw everything to a buffer first and then display it later  (when every element of the scene is drawn to buffer ) since else this happens: (it happens much faster when not recording it but it is still annoying to look at )
[image]

daniel-thompson, 2023-01-18:
The simulator is *hugely* inefficient at drawing. In fact it locks, copies and updates the display surface for every single pixel we draw. However this is entirely deliberate because it matches how the actual watches display hardware works. On modern hardware (at least with Intel graphics) the simulator is still likely to outperform the real hardware.

A pixel buffer simple cannot solve drawing performance problems once you try to run the application on the watch. There are a couple of reasons for this. Most importantly the pixel buffer does not fix into the RAM of the watch. However even it is did fit into RAM then the display would need ~125ms to copy the pixel buffer to the display (e.g. which limits you rame rate to 8Hz which isn't enough for a video game).

The only way to get acceptable video game performance on the watch hardware is by trickery... you have to design the graphical artifacts to avoid full screen draws. This can be a big limitation, especially for side scrolling since it essentially means the background must consist solely of horizonal lines (or solid color). For example you can't have triangular mountains in the background because you cannot side scroll these fast enough but you would do a space ship themed version of flappybird since a black background would make sense there.

After that you just have to use trickery to draw the pipes. You would to predraw a single pipe and encode it with rle_encode.py. You could then draw parts of that pipe as needed on the display. The trick is that each row would need to looks something like this:

DK MD BR BR BR BR MD DK BG BG
(DK -> dark, MD -> mid, BR -> bright, BG -> match background at current Y position)

You can now draw this pipe up to two pixels to the left of its previous location and the two BG pixels of overdraw have the effect of "moving" the pipe by erasing the old pixels and restoring the background.

wasp-os is full of trickery to hide draw times. Most noteably the GUI is very rectangular (since this is much faster to draw) and we usually disable the display (it goes black) whilst we are drawing the GUI so the user can't see the intermediate drawing states. Likewise you will notice that the heart rate application does not side scroll but updates the graph in place (this was the only way to get it to update fast enough to work in real time).

Note also that the ST7789 LCD controller hardware does implement vertical scrolling which could, in principle be used to create a fast vertically scrolling video game. Such a game would need to bypass wasp-os' built in drawing features and render directly to the hardware. However I'm afraid there are no examples of this to help and this feature is not implemented by in the simulator (because no software is using it). However in principle it is probably the best way to write a smooth game.

tigercoding56, 2023-01-19:
> The simulator is _hugely_ inefficient at drawing. In fact it locks, copies and updates the display surface for every single pixel we draw. However this is entirely deliberate because it matches how [...]

ok

closed 2023-01-19 by tigercoding56