wasp-os/wasp-os #396: [feature request] display heart rate on watchface

tigercoding56, 2023-02-18:
i would like the option to display the heart rate on a watchface along with the time (maybe sampling it every 2 minutes )

thiswillbeyourgithub, 2023-02-18:
I implemented this on my personnal waspos version [here](https://github.com/thiswillbeyourgithub/wasp-os)

But that's when I started doing this that I noticed that the heart rate was completely unreliable when moving the wrist or walking. Ultimately this lead to [this thread](https://github.com/wasp-os/wasp-os/issues/380) that is still pending resolution.

tigercoding56, 2023-02-18:
> I already implemented this on my personal waspos version [here](https://github.com/thiswillbeyourgithub/wasp-os)
> 
> But that's when I started doing this that I noticed that the heart rate was completely unreliable when moving the wrist or walking. Ultimately this lead to [this thread](https://github.com/wasp-os/wasp-os/issues/380) that is still pending resolution.

ok

closed 2023-02-18 by tigercoding56