wasp-os/wasp-os #407: [Build request] Build with apps such as SleepTk and more

tiPrans, 2023-03-12:
Hey everyone, apologies in advance for the dummy request! But would it be possible to provide the micropython.zip file of an os build containing some of the useful apps such as SleepTk, morse, music player and calculator? I've read through the documentation but couldn't build the os myself and couldn't find anywhere else to ask. Again, apologies for the dummy request!

tiPrans, 2023-03-14:
Thank you very much! However, I don't seem to be able to see Morse app in there. But SleepTk is in!

GaryM0101, 2023-03-14:
Sorry, yes morse and other larger applications like Level and Beacon are loaded via the wasptool onto the extended memory of the watch and are not included in the micropython.zip flash build. You'd need a Linux like machine and git clone the WASP_OS repository to upload the morse app and configure the local main.py file to run it.
Gary

tiPrans, 2023-03-14:
I see, thanks a lot Gary!

tiPrans, 2023-03-22:
hi again @GaryM0101 , I'd still like to upload a version of the OS that has some of my preferred apps. However, I am stuck at the process of compiling the OS and getting the micropython.zip file. I think the official tutorial might be incomplete in this regard. If you have the time, could you please share the steps/code required to get the micropython.zip? Sorry for the dummy request but I'm not sure where else to ask!

GaryM0101, 2023-03-23:
Hello @tiPrans I can build you another one with your preferred apps, note SleepTk has issues with the stepcounter and logging of it so I suggest not including it. Let me know your preferred apps and I can build it.

Though if you want to do this on your own and need help, it's better to meetup on Libera.chat https://app.element.io/#/room/#wasp-os:libera.chat

Gary

GaryM0101, 2023-03-23:
The Literal Clock errors when loading in the simulator:

File "/home/pi/wasp-os/wasp/apps/literal_clock.py", line 90, in update
    say = ("", hour[now[3] % 12], "fourty", lows[now[4]])
IndexError: list index out of range

So I used the Word Clock. Everything tested okay on the simulator but no guarantees since the simulator has unlimited memory. The Morse app tested okay too but not sure if it will work being frozen in the firmware.

[micropython.zip](https://github.com/wasp-os/wasp-os/files/11056978/micropython.zip)
 
Good Luck!
Gary

tiPrans, 2023-03-24:
thanks a lot Gary! That's so kind of you! I've tried it on the watch and everything works! I'll try to rework on the Literal Clock and let you know, maybe include the Sports app too if that's not too much to ask. 

btw, do you know if there is any to-do or note taking app for wasp-os? thanks!

GaryM0101, 2023-03-24:
@tiPrans Good! When you get the Literal Clock going I can make another build with the Sport App included.

As for a note taking app there is no "pull" request in the works but this is Python right? We could certainly write our own... ;-)

Gary

tiPrans, 2023-03-30:
Hi @GaryM0101 , would you be able to please try if this version of Literal Clock compiles ok when you have the time? Would it also be possible to please include the Sports and DisaBLE apps? Morse app can be removed if needed. Thanks!
[literal_clock_ENG.zip](https://github.com/wasp-os/wasp-os/files/11109820/literal_clock_ENG.zip)

tiPrans, 2023-03-31:
Hey @GaryM0101 sorry for spamming with yet another message but I've reworked the Literal Clock file and should be better. Could you please try to compile it along with the Sports and DisaBLE apps and let me know if there's any errors? Thanks!

[literal_clock_ENG.zip](https://github.com/wasp-os/wasp-os/files/11122107/literal_clock_ENG.zip)

GaryM0101, 2023-03-31:
@tiPrans
Looks like you have some more issues with the literal_clock, I tested this in the simulator.

File "/home/pi/wasp-os/wasp/apps/literal_clock.py", line 55, in foreground
    self.draw()
  File "/home/pi/wasp-os/wasp/apps/literal_clock.py", line 52, in draw
    self.update()
  File "/home/pi/wasp-os/wasp/apps/literal_clock.py", line 106, in update
    self.battery()
  File "/home/pi/wasp-os/wasp/apps/literal_clock.py", line 45, in battery
    draw.fill(c, 235, 239, 4, h)
  File "/home/pi/wasp-os/wasp/draw565.py", line 153, in fill
    display.set_window(x, y, w, h)
  File "/home/pi/wasp-os/wasp/drivers/st7789.py", line 139, in set_window
    window[2] = yp >> 8
TypeError: unsupported operand type(s) for >>: 'float' and 'int'

tiPrans, 2023-04-14:
Hey @GaryM0101 , I've spent some time and edited the word_clock.py file which is in the main OS' watch_faces folder. I think this should be less problematic to compile as it uses the default font. Could you please compile an OS with this as the sole watch face, include Sleep.tk and Sports app (Morse can be removed if needed).

Thanks a lot! 

[word_clock_MOD.zip](https://github.com/wasp-os/wasp-os/files/11236604/word_clock_MOD.zip)

GaryM0101, 2023-04-17:
@tiPrans 

The word clock compiled and tested okay in the simulator. I kept the other apps you requested before like calculator.py, music_player.py, play2048.py, snake.py, stopwatch.py, timer.py. Sleep.tk, DisaBLE and Sports apps are there.

[micropython.zip](https://github.com/wasp-os/wasp-os/files/11254681/micropython.zip)

tiPrans, 2023-04-17:
hey there, thanks a lot for your reply and for providing the compiled OS! I tried installing but after flashing the OS, the screen of the watch is stuck on 'main.py' :/ not sure if you know how to fix it but I'll look into it

GaryM0101, 2023-04-18:
I suspect we loaded too many apps and you're running out of memory. I recompiled with only Sleep.tk, DisaBLE and Sports app included. Give that a try and let me know.

[micropython.zip](https://github.com/wasp-os/wasp-os/files/11256742/micropython.zip)

tiPrans, 2023-04-21:
thanks! This one loads ok but there's an issue with the Sports app which doesn't seem to work well? Pressing the button when the app is open causes it to crash :/

thiswillbeyourgithub, 2023-04-21:
Hi! I'm the dev of SleepTk. I don't know which version of SleepTk you're using but one PR I made a while back has not been accepted so it might be a bit buggy I'm afraid: I made #370 because there are too many alarms and possibilities for me to keep track of it sanely so I prefered to modify wasp.cancel_alarm to be able to catch any matching time.

I'm pretty sure it's the only thing missing in terms of code for SleepTk to be working fine on the master branch.

Btw I have yet to upload the latest version that fixes the `natural wake` feature which is awesome for my sleep :) I'll try to get to fixing my #351 PR to make it mergeable.

GaryM0101, 2023-04-21:
There were a couple of python methods that were not implemented to use the current version, that's why we're using an older version that works with the main branch.

thiswillbeyourgithub, 2023-04-21:
I think you're refering to wasp.get and wasp.set. i added a check to call them only if present in wasp. Those methods are in #368 and i hope it gets merged someday.