wasp-os/wasp-os #409: Colmi P8 wasptool raise TIMEOUT('Timeout exceeded.') on --console

piersstorey, 2023-03-18:
After successfully uploading micropython.zip and connecting to GadgetBridge (See issue link [issues/402](https://github.com/wasp-os/wasp-os/issues/402) ) I've attempted to connect to the console and exec a file to the P8 using wasptool.

When checking memory with only the test app installed, the P8 only has  5856kb free

[image]

Console command used:
```
./tools/wasptool --console --verbose
```
Exec command used:
```
./tools/wasptool --exec myapp.py --eval "wasp.system.register(MyApp())" --verbose
```
On both commands I am seeing `raise TIMEOUT('Timeout exceeded.')`

Full trace
```
Connecting to P8b (DF:9A:6F:47:73:A6)...
Resolving services...
Exit console using Ctrl-X.
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 111, in expect_loop
    incoming = spawn.read_nonblocking(spawn.maxread, timeout)
  File "/usr/lib/python3/dist-packages/pexpect/pty_spawn.py", line 482, in read_nonblocking
    raise TIMEOUT('Timeout exceeded.')
pexpect.exceptions.TIMEOUT: Timeout exceeded.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./tools/wasptool", line 445, in <module>
    console.sync()
  File "./tools/wasptool", line 61, in sync
    c.expect('>>> ')
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 340, in expect
    return self.expect_list(compiled_pattern_list,
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 369, in expect_list
    return exp.expect_loop(timeout)
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 119, in expect_loop
    return self.timeout(e)
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 82, in timeout
    raise TIMEOUT(msg)
pexpect.exceptions.TIMEOUT: Timeout exceeded.
<pexpect.pty_spawn.spawn object at 0x7fc20d75cfd0>
command: ./tools/pynus/pynus.py
args: [b'./tools/pynus/pynus.py']
buffer (last 100 chars): '.\r\n'
before (last 100 chars): '.\r\n'
after: <class 'pexpect.exceptions.TIMEOUT'>
match: None
match_index: None
exitstatus: None
flag_eof: False
pid: 12843
child_fd: 5
closed: False
timeout: 30
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: <_io.TextIOWrapper name='<stdout>' mode='w' encoding='utf-8'>
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0.05
delayafterclose: 0.1
delayafterterminate: 0.1
searcher: searcher_re:
    0: re.compile('>>> ')
```

I've tried a couple of suggestions from [issues/23](https://github.com/wasp-os/wasp-os/issues/23)

@daniel-thompson  https://github.com/wasp-os/wasp-os/issues/23#issuecomment-631030986

> Try changing File "MicroPineTime/tools/pynus/tealblue.py", line 147 from:
> `if properties['ServicesResolved']:`
> to
> `if 'ServicesResolved' in properties:`

Also pulling the latest [wasp-os/pynus](https://github.com/wasp-os/pynus) branch

I'd value anyone's suggestions or support in how I can get passed the timeout issue 🙏

GaryM0101, 2023-03-18:
FWIW your pic above shows the P8 on the charger? I know having my Pinetime on the charger will cause wasptool timeouts, sporadic though. So when using wasptool I always keep the watch off the charger. Doubt that this is your cause but thought I'd mention it just in case.

piersstorey, 2023-03-18:
Thanks Gary. Unfortunately it's the same off charge ☹️.

I'm wondering if it's an available memory issue. I've seen a previous reference in one of the videos.

Only challenge is not being able to determine the free memory when the test app is disabled.

The Linux command times out as per the other commands..

Really hoping I can find a solution. The P8 seems decent enough wearable and I already have an app I'd like to develop that is already written for the Bangle.js smart watch 👍

GaryM0101, 2023-03-18:
You can try booting into safe mode, you can do a lot from here...https://wasp-os.readthedocs.io/en/latest/install.html#safe-mode

Try these:

pi@raspberrypi:~/wasp-os $ ./tools/wasptool --console
Connected to PineTime (FC:16:5F:3E:8E:7C).
Exit console using Ctrl-X.
from shell import*
>>> import wasp
>>> wasp.gc.mem_free()
12912

piersstorey, 2023-03-21:
Thanks, @GaryM0101,

In putting the P8 into Safe Mode

[image]

I've occasionally got a successful repl 🎉
```bash
./tools/wasptool --console
Connected to P8b (DF:9A:6F:47:73:A6).
Exit console using Ctrl-X.
from shell import*
>>> import wasp
>>> wasp.gc.mem_free()
11440
>>> 
```
The valuable output for me is understanding the available memory at 11440kb which should be acceptable for interfacing and uploading  👍

For the majority of attempts, I'm seeing the following stack trace

```bash
./tools/wasptool --console
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 166, in read_nonblocking
    s = os.read(self.child_fd, size)
OSError: [Errno 5] Input/output error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 111, in expect_loop
    incoming = spawn.read_nonblocking(spawn.maxread, timeout)
  File "/usr/lib/python3/dist-packages/pexpect/pty_spawn.py", line 485, in read_nonblocking
    return super(spawn, self).read_nonblocking(size)
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 171, in read_nonblocking
    raise EOF('End Of File (EOF). Exception style platform.')
pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./tools/wasptool", line 250, in <module>
    sync(console)
  File "./tools/wasptool", line 35, in sync
    c.expect('>>> ')
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 340, in expect
    return self.expect_list(compiled_pattern_list,
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 369, in expect_list
    return exp.expect_loop(timeout)
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 117, in expect_loop
    return self.eof(e)
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 63, in eof
    raise EOF(msg)
pexpect.exceptions.EOF: End Of File (EOF). Exception style platform.
<pexpect.pty_spawn.spawn object at 0x7f9e8842a580>
command: ./tools/pynus/pynus.py
args: [b'./tools/pynus/pynus.py']
buffer (last 100 chars): ''
before (last 100 chars): '.\r\nlost connection\r\n'
after: <class 'pexpect.exceptions.EOF'>
match: None
match_index: None
exitstatus: None
flag_eof: True
pid: 7212
child_fd: 5
closed: False
timeout: 30
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: None
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0.05
delayafterclose: 0.1
delayafterterminate: 0.1
searcher: searcher_re:
    0: re.compile('>>> ')
```
I would appreciate any thoughts or suggestions...

GaryM0101, 2023-03-21:
With that much memory I agree that you shouldn't have trouble getting to the REPL and be able to interface. One thing to check is your blue-tooth settings, ensuring nothing else is paired to it especially in your Linux gui. You might also un-pair the P8 from your phone and try again. Also keep the P8 near the blue-tooth source the wasptool is connecting to, as close as possible, even on my wrist near my pi4 I'll loose connection to wasptocd so I take it off and place it right next to the pi4. 

We also just released WASP al la Carte, you can now can easily make a custom build. https://wasp-os.readthedocs.io/en/latest/install.html#id14 But sounded like you loaded a minimal build in you previous post so I'm not so sure this will help you but an more f an FYI. 

Here is a scrrenshot from my gui. Note you can use "bluetoothctl" to monitor your connections. Note the gui blue-tooth list has no connections to remove and that's what you want to see. I hope this helps...Gary

[image]

piersstorey, 2023-03-21:
Thanks @GaryM0101,

So the issue turned out to be much simpler than expected.. Your line here:
> Also keep the P8 near the blue-tooth source the wasptool is connecting to, as close as possible, even on my wrist near my pi4 I'll loose connection to wasptool so I take it off and place it right next to the pi4.

was the primary issue. While the watch was probably no more than 60cm away from the laptop, the second I put the watch directly next to the laptop I was able to console in without being in Safe Mode.
```
Connected to P8b (DF:9A:6F:47:73:A6).
Exit console using Ctrl-X.
from shell import*
>>> import wasp
>>> wasp.gc.mem_free()
6832
>>> 
```
Register a simple app (Hence less memory now available)
```
./tools/wasptool --exec hello_app.py --eval "wasp.system.register(MyApp())" 
```
[image]

I'll try with a clean repo clone just to check but the console command has been tried multiple times and seems solid.

Thanks again for the support.

closed 2023-03-21 by piersstorey